dusty-nv/jetson-inference #1369: Long Post-Process time (due to slow Unified Memory read?)

voidzer0-development, 2022-02-21:
Hi dusy-nv / anyone reading,

I've been getting quite long Post-Process times while using my character recognizer model. See screenshot below.
[image]

My model has 37 classes (including background). I have determined the bulk of the post-processing time to be caused by a relatively slow read from UM on the following line:
https://github.com/dusty-nv/jetson-inference/blob/master/c/detectNet.cpp#L818

I have attempted copying the entire block over to cpu-only memory in one call and reading from that, which sped up the post process time to 6ms but only 1/4 of the pointers were valid for some reason.

Is there any way I can copy the entire managed memory page or run this part of the code in cuda (which should have faster UM access on the TX2 than the CPU if im not mistaken)? Any help or advice would be highly appreciated!

voidzer0-development, 2022-02-21:
The copy I've tried:
`uint32_t pageObjects = numBoxes * mNumClasses + mNumClasses;`
`void* cpuBlock  = malloc(pageObjects * 0x4);`
`CUDA(cudaMemcpy(cpuBlock, mOutputs[OUTPUT_CONF].CPU, pageObjects, cudaMemcpyDeviceToHost))`

Inside the 2 nested for loops:

`const float score = *(float*)(cpuBlock + ((n * mNumClasses + m) * 0x4));`

voidzer0-development, 2022-02-21:
Fixed by using `memcpy(cpuBlock,  mOutputs[OUTPUT_CONF].CPU, pageObjects * 0x4);`

[image]

closed 2022-02-21 by voidzer0-development

voidzer0-development, 2022-02-21:
@dusty-nv Did you consider implementing something like this? The performance increase seems to besignificant.

voidzer0-development, 2022-02-23:
I'm using the TX2 w/ JetPack 4.6 (rev.3) 

nvcc --version out:
`Cuda compilation tools, release 10.2, V10.2.300
Build cuda_10.2_r440.TC440_70.29663091_0
`

voidzer0-development, 2022-02-23:
If you need anything else please let me know i'm happy to provide it to you.

voidzer0-development, 2022-02-23:
I think the issue here is that mapped memory on the jetson tx2 doesn't have the hardware support to sync with the CPU cache unlike other pascal based devices. But I might be mistaken.

dusty-nv, 2022-02-23:
> I think the issue here is that mapped memory on the jetson tx2 doesn't have the hardware support to sync with the CPU cache unlike other pascal based devices.

Hmm I think you are probably onto something.   Xavier and newer is cache-coherent for mapped memory.

voidzer0-development, 2022-02-24:
Is there a way to automatically detect if the program is being ran on a device that supports CPU cache coherence, or at least the architecture (pascal/volta etc)? I'd like to run these optimizations only when necessary.

voidzer0-development, 2022-02-24:
I forked the repository here: https://github.com/voidzer0-development/jetson-inference-TX2-optimized
Will PR once I implement the detection described above

voidzer0-development, 2022-03-01:
Maybe the reason you haven't seen this issue yet is that the default nets have a low amount of max boxes (for example 100 in mobilenetv2) and the network im using (VGG16SSD) has 8372.