dusty-nv/jetson-inference #1372: Getting Classes from Training Test

eanmikale, 2022-02-23:
Dusty, 
During training, it would be extremely useful if we can see which classes are underperforming during testing, so we may more efficiently adjust data-sets between training. Is there a way to do this with the jetson.utils library? Thank you.

dusty-nv, 2022-02-23:
Hi @eanmikale, that wouldn't be a jetson.inference/jetson.utils feature, because those libraries are used for runtime inferencing only.   This would be something to be added to the PyTorch training scripts.  In theory it's possible, but I haven't looked into it.  I get the PyTorch training scripts from other sources.

eanmikale, 2022-02-24:
For anyone else who might be interested, it seems this might be the closest we can gethttps://stackoverflow.com/questions/62958248/calculate-accuracy-for-each-class-using-cnn-and-pytorch. We can close it as far as I am concerned. Thanks Dusty.  ---- On Wed, 23 Feb 2022 12:48:36 -0600  ***@***.***  wrote ----
Hi @eanmikale, that wouldn't be a jetson.inference/jetson.utils feature, because those libraries are used for runtime inferencing only.   This would be something to be added to the PyTorch training scripts.  In theory it's possible, but I haven't looked into it.  I get the PyTorch training scripts from other sources.

closed 2023-04-27 by dusty-nv