dusty-nv/jetson-inference #1373: Skipping cmake interactive pop-ups (automated script)

firststep-dev, 2022-02-24:
Is there a way to run `cmake ../` step (as part of an automated script), automatically skipping these 2 pop-ups?

I am running the following commands:

```
rm -rf build
mkdir build
cd build
cmake ../
make -j$(nproc)
sudo make install
```

Step 1: Quit

[image]

Step 2: Skip

[image]

dusty-nv, 2022-02-24:
Hi @firststep-dev, yes there is, with [`-DBUILD_INTERACTIVE=off`](https://github.com/dusty-nv/jetson-inference/blob/3a032b0992d7786b00bcbf19232a8822b6160cf9/CMakeLists.txt#L28)

firststep-dev, 2022-02-25:
Thanks @dusty-nv

firststep-dev, 2022-02-27:
@dusty-nv setting this `-DBUILD_INTERACTIVE=off` flag by default this installs "default models".

```
Reading package lists...
[jetson-inference]  BUILD_INTERACTVE=off
[jetson-inference]  Downloading default models...
[jetson-inference]  Downloading GoogleNet...
bvlc_googlenet.caff   3%[                    ]   2,00M   122KB/s    eta 5m 34s
```

How can I disable / skip these 2 pop-ups (i.e. edit the default lists) for non-interactive build mode?

leenremm, 2022-04-27:
@dusty-nv - just a kind followup.

How can I run `-DBUILD_INTERACTIVE=off` while disabling ALL default model download? 

Is there a config file for default model downloads?

leenremm, 2022-04-27:
I see that the default list of models is defined in a SH script in `build/download-models.sh`:

```
if [[ "$BUILD_INTERACTIVE" != "YES" ]]; then

        echo "$LOG Downloading default models..."

        download_googlenet
        download_resnet18

        download_ssd_mobilenet_v2
        download_pednet
        download_facenet
        download_detectnet_coco_dog

        download_fcn_resnet18_cityscapes_512x256
        download_fcn_resnet18_cityscapes_1024x512
        download_fcn_resnet18_deepscene_576x320
        download_fcn_resnet18_mhp_512x320
        download_fcn_resnet18_pascal_voc_320x320
        download_fcn_resnet18_sun_rgbd_512x400

        exit_message 0
fi

```

leenremm, 2022-04-27:
I see that this file is created in the `./build` directory during `cmake`, where it is copied from `tools/download-models.sh`

file(COPY "tools/download-models.sh" DESTINATION ${PROJECT_BINARY_DIR})

So I have edited the file: `tools/download-models.sh`, from:

```
if [[ "$BUILD_INTERACTIVE" != "YES" ]]; then

	echo "$LOG Downloading default models..."

        exit_message 0

	download_googlenet
	download_resnet18

	download_ssd_mobilenet_v2
	download_pednet
	download_facenet
	download_detectnet_coco_dog

	download_fcn_resnet18_cityscapes_512x256
	download_fcn_resnet18_cityscapes_1024x512
	download_fcn_resnet18_deepscene_576x320
	download_fcn_resnet18_mhp_512x320
	download_fcn_resnet18_pascal_voc_320x320
	download_fcn_resnet18_sun_rgbd_512x400

	exit_message 0
fi
```

to the following:

```
if [[ "$BUILD_INTERACTIVE" != "YES" ]]; then

	echo "$LOG Downloading default models..."
        exit_message 0

fi
```

ligaz, 2022-04-27:
Alternatively you can remove the download step in the pre-build script: [`CMakePreBuild.sh:46`](https://github.com/dusty-nv/jetson-inference/blob/6cf3b12f503d64903ca1e77d4c5474d9b1513a4c/CMakePreBuild.sh#L46).

dusty-nv, 2022-04-27:
Hi @ligaz, very good, glad you got it figured out - I would have suggested either of your patches.

closed 2023-04-27 by dusty-nv