dusty-nv/jetson-inference #1376: NotADirectoryError: [Errno 20] Not a directory:

ltephanysopez, 2022-02-28:


dusty-nv, 2022-02-28:
Hi @ltephanysopez, the `--model-dir` argument expects a directory, not the full path to your pth checkpoint.  Can you try running it as `python3 onnx_export.py --model-dir=models/euler` instead?

If you are trying to specify a specific pth checkpoint to use (by default, the one with the lowest loss is selected), then use the `--input` and `--output` arguments:  https://github.com/dusty-nv/pytorch-ssd/blob/3f9ba554e33260c8c493a927d7c4fdaa3f388e72/onnx_export.py#L20

ltephanysopez, 2022-03-01:
Thanks, Dustin! That fixed the error :) On an unrelated note, I was wondering if it was possible to train with multiple datasets in and outside of a Docker container? I'm currently trying to migrate data on my host (roughly 37k pictures/annotations) into my data directory in the container but it's not as easy of a process as I thought. Any insight here would be appreciated 🙏🏽 

@dusty-nv

dusty-nv, 2022-03-01:
> 'm currently trying to migrate data on my host (roughly 37k pictures/annotations) into my data directory in the container but it's not as easy of a process as I thought.

Oh sure - when you start the container, you can mount an additional path into the container using the `--volume` flag:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md#mounted-data-volumes

So just mount your folder containing your dataset, and then point train_ssd.py to that mounted path instead.

ltephanysopez, 2022-03-03:
got it- thank you so much @dusty-nv! :)

closed 2022-03-03 by ltephanysopez