dusty-nv/jetson-inference #1377: Which API call one has to use? to input a video from a file to further do object detection

snsmssss, 2022-03-01:
In the my-detection.py  example part of object detection/jetson inference tutorial, it is shown how to use
camera feed as follows:
camera = jetson.utils.gstCamera(1280,720,"/dev/video0")

Here, instead of camera source, I want to use a video from a file say (Car.mp4)
Appreciate guidance on 
Which API call one has to use? 
I tried jetson.utils.videoSource()
Some times I got openGL error, some times segmentation fault.

My aim is to take a  frame with a particular object of interest , out of the video file and investigate further ...

dusty-nv, 2022-03-01:
Hi @snsmssss, please see the updated tutorial here that is using the videoSource/videoOutput APIs:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-example-2.md
https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/my-detection.py

Or you can just use `detectnet.py` and run it from video file like here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-console-2.md#processing-video-files

There is additional documentation on the streaming and video API's here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md

snsmssss, 2022-03-02:
Hello Dusty
After many trials, my mp4 video is playing , but object detection is wrong
For some reason my video of a car and person near the car gets
rotated anti clock wise 90 deg
That may be the reason why the objects are not detected?
I use detectnet.py without change
I use mp4 video shot from my iphone

I got many alerts as given below

nvbuf_utils: dmabuf_fd 1299 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...
nvbuf_utils: dmabuf_fd 1301 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...
nvbuf_utils: dmabuf_fd 1303 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...

I also got memory shortage warning

dusty-nv, 2022-03-02:
Can you try running it with `--input-flip=rotate-180` ?

See here for the flags:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options

snsmssss, 2022-03-03:
Thank you for all your pertinent inputs. I am able to get what I want. No
memory shortage issues also now

I want to detect persons only ( my previous Human detection code based on
TF worked on my GPU server but had severe resource constraints on Nano) so
your tutorial exactly met my requirement. Now that the code is running on
Nano (with the help of TRT )

One last thing ...
As  soon as the person is detected by the  detector from a camera or .mp4
file ,  I want to save that frame only in the disk as, say, person.jpg
containing the person with the bounding box.

To achieve this, I have been trying the following but not working

...
for detection in detections:
                if detection.ClassID==1:
                        output.Render(img)
                break
...

Any suggestions ? Is it possible to disable other objects detection with
SSDMobilenet? Even simple human detection running optimally on nano will
help

dusty-nv, 2022-03-03:
> As soon as the person is detected by the detector from a camera or .mp4 file , I want to save that frame only in the disk as, say, person.jpg containing the person with the bounding box.

Hi @snsmssss, see this example script which saves a snapshot of the detected objects:   https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet-snap.py

snsmssss, 2022-03-04:
Thank you ! detectnet-snap is useful.

my next step is to upload the file to cloud

On NANO even after installing boto3,

the
import boto3
fails

I am not using the container. I am running detectnet on the non container
version

snsmssss, 2022-03-04:
Hello Dustin

I used the python3 version. Now it is resolved
Pls ignore the boto3 question

snsmssss, 2022-03-10:
Hello Dusty

I have put together a Burglary Prevention application using /customizing
detectnet-snap.py code

I want to submit this as an solution / blog for AI-Jetson-nano certification

I will be acknowledging NVIDIA and you for all the support

Can I share the modified code in github - with proper ack.

Ca*n I do ? Do I need to take care of anything else ?*

*Thank you for your guidance*

dusty-nv, 2022-03-10:
Sure no problem, you can put it on GitHub

<https://bit.ly/3DbGveC>

-