dusty-nv/jetson-inference #1378: No optimization profile defined

EulersBoiler, 2022-03-01:
I would like to load an .onnx of a segmentation network i trained to run on  a Jetson Nano 2GB. When loading the network with jetson.inference.net() the engine is started to build, but it fails due to a missing optimization profile. I saw a tag --profile to pass a profile as an argument, but i do not see any references to how i create and save an optimization profile. 

I saw the tutorial on dynamic shapes https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#work_dynamic_shapes, but not how to save and load such a profile

Thanks for your help

dusty-nv, 2022-03-01:
Hi @EulersBoiler, what segmentation model are you using?   `jetson.inference.segNet` is setup to use FCN-ResNet and FCN-Alexnet.  FCN-Alexnet is legacy model from caffe (no longer used for new models really), whereas FCN-ResNet is exported from PyTorch to ONNX with fixed shapes, so it doesn't need the dynamic shape profiles.

For a tutorial on training FCN-ResNet with PyTorch, see here:   https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson

EulersBoiler, 2022-03-02:
I further looked into my model and cant seem to find any dynamic parts. It all seems to have known values. Down below the graph printed by the onnx module for python

```
graph Network (
  %data[FLOAT, BatchSizex3x224x224]
) initializers (
  %data_Mean[FLOAT, 1x3x1x1]
  %data_StandardDeviation[FLOAT, 1x3x1x1]
  %conv1_W[FLOAT, 64x3x7x7]
  %conv1_B[FLOAT, 64]
  %bn_conv1_scale[FLOAT, 64]
  %bn_conv1_B[FLOAT, 64]
  %bn_conv1_mean[FLOAT, 64]
  %bn_conv1_var[FLOAT, 64]
  %res2a_branch2a_W[FLOAT, 64x64x3x3]
  %res2a_branch2a_B[FLOAT, 64]
  %bn2a_branch2a_scale[FLOAT, 64]
  %bn2a_branch2a_B[FLOAT, 64]
  %bn2a_branch2a_mean[FLOAT, 64]
  %bn2a_branch2a_var[FLOAT, 64]
  %res2a_branch2b_W[FLOAT, 64x64x3x3]
  %res2a_branch2b_B[FLOAT, 64]
  %bn2a_branch2b_scale[FLOAT, 64]
  %bn2a_branch2b_B[FLOAT, 64]
  %bn2a_branch2b_mean[FLOAT, 64]
  %bn2a_branch2b_var[FLOAT, 64]
  %res2b_branch2a_W[FLOAT, 64x64x3x3]
  %res2b_branch2a_B[FLOAT, 64]
  %bn2b_branch2a_scale[FLOAT, 64]
  %bn2b_branch2a_B[FLOAT, 64]
  %bn2b_branch2a_mean[FLOAT, 64]
  %bn2b_branch2a_var[FLOAT, 64]
  %res2b_branch2b_W[FLOAT, 64x64x3x3]
  %res2b_branch2b_B[FLOAT, 64]
  %bn2b_branch2b_scale[FLOAT, 64]
  %bn2b_branch2b_B[FLOAT, 64]
  %bn2b_branch2b_mean[FLOAT, 64]
  %bn2b_branch2b_var[FLOAT, 64]
  %res3a_branch2a_W[FLOAT, 128x64x3x3]
  %res3a_branch2a_B[FLOAT, 128]
  %bn3a_branch2a_scale[FLOAT, 128]
  %bn3a_branch2a_B[FLOAT, 128]
  %bn3a_branch2a_mean[FLOAT, 128]
  %bn3a_branch2a_var[FLOAT, 128]
  %res3a_branch2b_W[FLOAT, 128x128x3x3]
  %res3a_branch2b_B[FLOAT, 128]
  %bn3a_branch2b_scale[FLOAT, 128]
  %bn3a_branch2b_B[FLOAT, 128]
  %bn3a_branch2b_mean[FLOAT, 128]
  %bn3a_branch2b_var[FLOAT, 128]
  %res3a_branch1_W[FLOAT, 128x64x1x1]
  %res3a_branch1_B[FLOAT, 128]
  %bn3a_branch1_scale[FLOAT, 128]
  %bn3a_branch1_B[FLOAT, 128]
  %bn3a_branch1_mean[FLOAT, 128]
  %bn3a_branch1_var[FLOAT, 128]
  %res3b_branch2a_W[FLOAT, 128x128x3x3]
  %res3b_branch2a_B[FLOAT, 128]
  %bn3b_branch2a_scale[FLOAT, 128]
  %bn3b_branch2a_B[FLOAT, 128]
  %bn3b_branch2a_mean[FLOAT, 128]
  %bn3b_branch2a_var[FLOAT, 128]
  %res3b_branch2b_W[FLOAT, 128x128x3x3]
  %res3b_branch2b_B[FLOAT, 128]
  %bn3b_branch2b_scale[FLOAT, 128]
  %bn3b_branch2b_B[FLOAT, 128]
  %bn3b_branch2b_mean[FLOAT, 128]
  %bn3b_branch2b_var[FLOAT, 128]
  %res4a_branch2a_W[FLOAT, 256x128x3x3]
  %res4a_branch2a_B[FLOAT, 256]
[... 193 lines cut ...]
  %catAspp = Concat[axis = 1](%aspp_Relu_1, %aspp_Relu_2, %aspp_Relu_3, %aspp_Relu_4)
  %dec_c1 = Conv[dilations = [1, 1], group = 1, kernel_shape = [1, 1], pads = [0, 0, 0, 0], strides = [1, 1]](%catAspp, %dec_c1_W, %dec_c1_B)
  %dec_bn1 = BatchNormalization[epsilon = 9.99999974737875e-06](%dec_c1, %dec_bn1_scale, %dec_bn1_B, %dec_bn1_mean, %dec_bn1_var)
  %dec_relu1 = Relu(%dec_bn1)
  %dec_upsample1 = ConvTranspose[dilations = [1, 1], group = 1, kernel_shape = [8, 8], pads = [2, 2, 2, 2], strides = [4, 4]](%dec_relu1, %dec_upsample1_W, %dec_upsample1_B)
  %dec_c2 = Conv[dilations = [1, 1], group = 1, kernel_shape = [1, 1], pads = [0, 0, 0, 0], strides = [1, 1]](%res2b_relu, %dec_c2_W, %dec_c2_B)
  %dec_bn2 = BatchNormalization[epsilon = 9.99999974737875e-06](%dec_c2, %dec_bn2_scale, %dec_bn2_B, %dec_bn2_mean, %dec_bn2_var)
  %dec_relu2 = Relu(%dec_bn2)
  %dec_crop1 = Slice[ends = [2147483647, 2147483647, 2147483647, 2147483647], starts = [0, 0, 0, 0]](%dec_upsample1)
  %dec_cat1 = Concat[axis = 1](%dec_relu2, %dec_crop1)
  %dec_c3 = Conv[dilations = [1, 1], group = 1, kernel_shape = [3, 3], pads = [1, 1, 1, 1], strides = [1, 1]](%dec_cat1, %dec_c3_W, %dec_c3_B)
  %dec_bn3 = BatchNormalization[epsilon = 9.99999974737875e-06](%dec_c3, %dec_bn3_scale, %dec_bn3_B, %dec_bn3_mean, %dec_bn3_var)
  %dec_relu3 = Relu(%dec_bn3)
  %dec_c4 = Conv[dilations = [1, 1], group = 1, kernel_shape = [3, 3], pads = [1, 1, 1, 1], strides = [1, 1]](%dec_relu3, %dec_c4_W, %dec_c4_B)
  %dec_bn4 = BatchNormalization[epsilon = 9.99999974737875e-06](%dec_c4, %dec_bn4_scale, %dec_bn4_B, %dec_bn4_mean, %dec_bn4_var)
  %dec_relu4 = Relu(%dec_bn4)
  %scorer = Conv[dilations = [1, 1], group = 1, kernel_shape = [1, 1], pads = [0, 0, 0, 0], strides = [1, 1]](%dec_relu4, %scorer_W, %scorer_B)
  %dec_upsample2 = ConvTranspose[dilations = [1, 1], group = 1, kernel_shape = [8, 8], pads = [2, 2, 2, 2], strides = [4, 4]](%scorer, %dec_upsample2_W, %dec_upsample2_B)
  %dec_crop2 = Slice[ends = [2147483647, 2147483647, 2147483647, 2147483647], starts = [0, 0, 0, 0]](%dec_upsample2)
  %softmax_out_Transpose1 = Transpose[perm = [0, 2, 3, 1]](%dec_crop2)
  %softmax_out = Softmax[axis = 3](%softmax_out_Transpose1)
  %softmax_out_Transpose2 = Transpose[perm = [0, 3, 1, 2]](%softmax_out)
  return %softmax_out_Transpose2
}
```

dusty-nv, 2022-03-02:
Hi @EulersBoiler, I'm not sure about your model, but the way that I export these to ONNX from PyTorch, it is with a fixed input size:    https://github.com/dusty-nv/pytorch-segmentation/blob/6ae33ba3426665b8bde8923c0434d207ea4331e0/onnx_export.py#L54

EulersBoiler, 2022-03-03:
Alright so i found a solution. I export my models from Matlab which makes two differences to your models in the first layer that one can corrected with the onnx module for python.

EulersBoiler, 2022-03-03:
For completeness sake here the solution. The problem is that the onnx is exported with a BATHSIZE parameter in the first layer from which all consecutive layers inherit, therby being dynamic. The first layer has 4 dimensions, i.e. BATCHSIZEx3x224x224 for RGB segmentation. Changing the first dimension to 1 however does not solve the problem, as the dimension is a dim_param and not a dim_value as the other three dimensions. The solution is therefore the one below.
```
import onnx

model = onnx.load( onnxFilePath )
model.graph.input[0].type.tensor_type.shape.dim[0].dim_value = 1
onnx.save(model, targetFilePath)

```

closed 2023-04-27 by dusty-nv