dusty-nv/jetson-inference #1379: There are problems even though I used 100 epochs models

SangkyunAhn, 2022-03-02:
Hi.
I use Jetson Nano.

I tried these examples,
 - Re-training on the Cat/Dog Dataset(Classification, Recognition)
 - Re-training SSD-Mobilenet(Object Detection)
But, even though I used 100 epochs models, the problems are occurred.

In case of classification, the following problem is occurred.
[image]
[image]
Not only percentage is too low, but also Jetson recognizes dog as cat.

In case of object detection, some fruits are detected, but the others are not detected. But the fruits that are detected don't always detect accurately, for example, some banana pictures are detected, but others are not detected.

Thank you for reading this issue.

dusty-nv, 2022-03-02:
Hi @SangkyunAhn, can you try deleting the `*.engine` files from your model's directories and try running imagenet/detectnet again?

Also, you can find my trained models for cat/dog and the fruits dataset on the tutorial pages.  Can you try those to see if there is a difference?

SangkyunAhn, 2022-03-03:
I solved these problems. Thank you for your help.

closed 2023-04-27 by dusty-nv