dusty-nv/jetson-inference #1382: Example of TensorRT SSD and Pytorch

rsamvelyan, 2022-03-10:
Hello

I need much more flexibility as far as the window with the webcam feed is concerned. I need to use OpenCV in order to draw my custom boxes and labels. Something that the jetson-inference doesn't allow me to do.

Where could I get examples of how to integrate OpenCV webcam with Pytorch SSD inference and TensorRT optimization?

In other words I'm looking for a reference that would allow me to re-create the same thing as jetson-inference does but in Python.

Thanks!

dusty-nv, 2022-03-10:
Hi @rsamvelyan, you can use the `cudaFromNumpy()` and `cudaToNumpy()` functions from jetson.utils to convert images to/from OpenCV:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays

* https://github.com/dusty-nv/jetson-utils/blob/7de695d94271ba1ae77a4358f427c0887b09084f/python/examples/cuda-from-cv.py
* https://github.com/dusty-nv/jetson-utils/blob/7de695d94271ba1ae77a4358f427c0887b09084f/python/examples/cuda-to-cv.py

rsamvelyan, 2022-03-10:
Ok great! Will give it a try.
I have my custom trained PTH files which work very well with jetson-inference but so far I need to convert them to ONNX.
Is there a way to run inference on the .pth files directly instead of heaving to convert them to .onnx?

dusty-nv, 2022-03-10:
> Is there a way to run inference on the .pth files directly instead of heaving to convert them to .onnx?

No, because jetson-inference runtime library doesn't use PyTorch - it uses TensorRT, and TensorRT imports ONNX

rsamvelyan, 2022-03-10:
OK I see. And if I wanted to use directly PHT files then I would have to write my own inference from scratch?

dusty-nv, 2022-03-10:
> And if I wanted to use directly PHT files then I would have to write my own inference from scratch?

You would just use PyTorch to run PTH checkpoints...

rsamvelyan, 2022-03-10:
And in that case would not be able to optimize it with TensroRT, correct?

dusty-nv, 2022-03-10:
Correct, although you can try use torch2trt which uses TensorRT and doesn't need ONNX:   https://github.com/NVIDIA-AI-IOT/torch2trt

rsamvelyan, 2022-03-10:
Thanks as always Dusty! Really appreciate your help. My projects would probably have stalled without your help!!

dusty-nv, 2022-03-10:
No problem @rsamvelyan, happy to help!  Wish you the best of luck on your project!

rsamvelyan, 2022-03-15:
I created the live webcam capture that uses OPEN CV. It works very well and the FPS is very high - over 30 FPS on Jetson Nano. I can share it if you want.

closed 2022-03-21 by rsamvelyan

hungpham3112, 2025-04-10:
> I created the live webcam capture that uses OPEN CV. It works very well and the FPS is very high - over 30 FPS on Jetson Nano. I can share it if you want.

Can I see your implementation details? I am struggling with the part of converting SSDLite320 to tensorRT