dusty-nv/jetson-inference #1385: TensorRT Optimization warning message

rsamvelyan, 2022-03-11:
Hello

I launched an inference on my custom mobilenetv1 ssd and during the optimization I got this message in yellow.

Am I doing something wrong? Did I miss specifying something? 

Thank you!

**[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8**

import cv2
import jetson.inference
import jetson.utils

import argparse
import sys

# print(cv2.getCudaEnabledDeviceCount())

# parse the command line
parser = argparse.ArgumentParser(description="Locate objects in a live camera stream using an object detection DNN.", 
						   formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.detectNet.Usage())
parser.add_argument("--network", type=str, default="ssd-mobilenet-v1", help="pre-trained model to load (see below for options)")
parser.add_argument("--overlay", type=str, default="box,labels,conf", help="detection overlay flags (e.g. --overlay=box,labels,conf)\nvalid combinations are:  'box', 'labels', 'conf', 'none'")
parser.add_argument("--threshold", type=float, default=0.8, help="minimum detection threshold to use") 
parser.add_argument("--camera", type=str, default="/dev/video0", help="index of the MIPI CSI camera to use (e.g. CSI camera 0)\nor for VL42 cameras, the /dev/video device to use.\nby default, MIPI CSI camera 0 will be used.")
parser.add_argument("--width", type=int, default=640, help="desired width of camera stream (default is 1280 pixels)")
parser.add_argument("--height", type=int, default=480, help="desired height of camera stream (default is 720 pixels)")

arguments  =   ['--model=/home/ryan/My_Projects/sniper/object_detection/mb1-ssd-apple-100.onnx', \
				'--class_labels=/home/ryan/My_Projects/sniper/object_detection/open-images-model-labels.txt', \
				'--input_blob=input_0', \
				'--output_cvg=scores', \
				'--output_bbox=boxes', \
				'--camera=/dev/video0', \
				'--width=640', \
				'--hight=480', \
                '--threshold=0.8']

try:
    opt = parser.parse_known_args()[0]

    # load the object detection network
    net = jetson.inference.detectNet(opt.network, arguments, opt.threshold)

    # create the camera and display
    camera = jetson.utils.gstCamera(opt.width, opt.height, opt.camera)
    display = jetson.utils.glDisplay(width=camera.GetWidth()+10, height=camera.GetHeight()+50)

except:
    print("exception thrown")
    parser.print_help()
    sys.exit(0)

while(True):

    # capture the image
    img, width, height = camera.CaptureRGBA()

    # detect objects in the image (with overlay)
    detections = net.Detect(img, width, height, opt.overlay)

    # render the image
    display.RenderOnce(img, width, height)   

    # update the title bar
    display.SetTitle("{:s} | Network {:.0f} FPS".format(opt.network, net.GetNetworkFPS()))

dusty-nv, 2022-03-11:
> [TRT] requested fasted precision for device GPU without providing valid calibrator, disabling INT8

Hi @rsamvelyan, no worries - you can safely ignore that

rsamvelyan, 2022-03-11:
Thanks a lot!

closed 2022-03-11 by rsamvelyan