dusty-nv/jetson-inference #1388: [TRT]    poseNet::postProcess() -- duplicate link detected, skipping...

AK51, 2022-03-17:
Hi,

I try to use "jetson.utils.cudaCrop" but it gives me these warning, will it affect the speed? Thx
I just want to crop certain area  and do the poseNet
Here is the warning [TRT]    poseNet::postProcess() -- duplicate link detected, skipping...
```
#Frame Start&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
Start poseNet 
[TRT]    poseNet::postProcess() -- duplicate link detected, skipping...
[TRT]    poseNet::postProcess() -- duplicate link detected, skipping...
len(poses) =  1
```

Below is my simplified code.
```
#!/usr/bin/python3
#
# Copyright (c) 2021, NVIDIA CORPORATION. All rights reserved.
#
# Permission is hereby granted, free of charge, to any person obtaining a
# copy of this software and associated documentation files (the "Software"),
# to deal in the Software without restriction, including without limitation
# the rights to use, copy, modify, merge, publish, distribute, sublicense,
# and/or sell copies of the Software, and to permit persons to whom the
# Software is furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.  IN NO EVENT SHALL
# THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
# FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
# DEALINGS IN THE SOFTWARE.
#
import jetson.inference
import jetson.utils
import argparse
import sys

# parse the command line
parser = argparse.ArgumentParser(description="Run pose estimation DNN on a video/image stream.", 
                                 formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.poseNet.Usage() +
                                 jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("input_URI", type=str, default="", nargs='?', help="URI of the input stream")
parser.add_argument("output_URI", type=str, default="", nargs='?', help="URI of the output stream")
parser.add_argument("--network", type=str, default="densenet121-body", help="pre-trained model to load (see below for options)")
parser.add_argument("--overlay", type=str, default="none", help="pose overlay flags (e.g. --overlay=links,keypoints)\nvalid combinations are:  'links', 'keypoints', 'boxes', 'none'")
parser.add_argument("--threshold", type=float, default=0.15, help="minimum detection threshold to use") 

try:
	opt = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)

# load the pose estimation model
net = jetson.inference.poseNet(opt.network, sys.argv, opt.threshold)

# create video sources & outputs
output = jetson.utils.videoOutput(opt.output_URI, argv=sys.argv)
input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)

img = input.Capture()
cuda_mem = jetson.utils.cudaAllocMapped(width=img.width,
                                         height=img.height,
                                         format=img.format)
while True:
    print("#Frame Start&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&")
    img = input.Capture()
    crop_roi = (0, 0, img.width, img.height)
    jetson.utils.cudaCrop(img, cuda_mem, crop_roi)
    print("Start poseNet ")
    poses = net.Process(cuda_mem, overlay="none")
    print("len(poses) = ", len(poses))
    output.Render(img)
    if not input.IsStreaming() or not output.IsStreaming():
        break
```

dusty-nv, 2022-03-17:
> I try to use "jetson.utils.cudaCrop" but it gives me these warning, will it affect the speed?

Hi @AK51, no it won't, you can safely ignore them

To supress them, you can run with `--log-level=error` or comment out this line of code and rebuild/reinstall:
https://github.com/dusty-nv/jetson-inference/blob/01a395892ecc8acdbec4d8e9d6e8ac676416a507/c/poseNet.cpp#L625

AK51, 2022-03-18:
Good to know it won't affect the speed. Thx Dusty. :>

closed 2022-03-18 by AK51

eyast, 2023-05-22:
I would like to repress these messages as well. I am not sure which argument to give, to which object. I've tried added '--log-level':'error' to the args dictionary, and passing this to the pose inference function but both failed. In my case, I am not cropping an Image - I am just running poseNet directly on the whole image. I'd appreciate some help - thanks!

dusty-nv, 2023-05-23:
@eyast try adding this to your script:

``` bash
from jetson_utils import Log
Log.SetLevel('error')
```

If that doesn't work, comment out this line:   https://github.com/dusty-nv/jetson-inference/blob/e1ee5676bdffc68f18569612403bc147cf9ae585/c/poseNet.cpp#L635
and re-run `make && sudo make install`