dusty-nv/jetson-inference #1389: Docker refuses to run

xShockey, 2022-03-20:
step 1: I clone repo
step 2: cd jetson-inference
step 3: docker/run.sh

step 4: terminal prints "
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.6.1
"
then it does nothing just lets my cursor blink with no directory n stuff, i can type still
how do i make the docker go boom and download stuff and open itself? (tried sudo)
im new to this stuff but if you wanna be mean go for it

xShockey, 2022-03-22:
There's only 1 terminal instance opened
[image]
[image]
, here's how it looks like, it seems like it's thinking but even after an hour it achieves nothing. 
```sudo docker pull dustynv/jetson-inference:r32.6.1``` also hangs

larryw-jimi, 2022-06-12:
For my trial, sudo docker pull dustynv/jetson-inference:r32.7.1 can run smoothly, but docker/run.sh will hang at [jetson-inference] Downloading GoogleNet... for hours, are there any other methods to download all these models ?

larryw-jimi, 2022-06-12:
Here are the detail info after turn on --verbose 

etson-inference]  Downloading GoogleNet...
--2022-06-12 10:56:49--  https://nvidia.box.com/shared/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel
Resolving nvidia.box.com (nvidia.box.com)... 103.116.4.197
Connecting to nvidia.box.com (nvidia.box.com)|103.116.4.197|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: /public/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel [following]
--2022-06-12 10:56:49--  https://nvidia.box.com/public/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel
Reusing existing connection to nvidia.box.com:443.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://nvidia.app.box.com/public/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel [following]
--2022-06-12 10:56:50--  https://nvidia.app.box.com/public/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel
Resolving nvidia.app.box.com (nvidia.app.box.com)... 203.98.7.65, 2001::adfc:5903
Connecting to nvidia.app.box.com (nvidia.app.box.com)|203.98.7.65|:443...

dusty-nv, 2022-06-13:
Hi @larryw-jimi, yes you can download the models from github mirror here:  https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618

closed 2023-04-27 by dusty-nv