dusty-nv/jetson-inference #1393: No module named torch.fx

FebrianFrancione, 2022-03-27:
Hi!
I'm  trying to train my custom data my own ssd model from custom dataset but when I try to train, I get no module named torch.fx.
How can I resolve this issue?

Thank you!

FebrianFrancione, 2022-03-30:
> Hi @FebrianFrancione <https://github.com/FebrianFrancione>, are you using
> train_ssd.py or your own PyTorch code? I'm unaware of train_ssd.py needing
> torch.fx module. Also which version of PyTorch are you running, and did you
> install one of my wheels or containers?

Hi Dustin,
Thank you for your response.
It might be an issue with the pytorch version actually. I'm on 1.6.0.
Apparently torch.fx was added in pytorch 1.8.0. Which version am I
supposed  to be on?

Thanks again!

FebrianFrancione, 2022-03-30:
Hi again, just an update. This issue was resolved when I updated pytorch :)
I have a question  though, it seems my training is not using CUDA but I
have CUDA set up. How can I make sure it runs on CUDA?

Thank you!

dusty-nv, 2022-03-30:
> It might be an issue with the pytorch version actually. I'm on 1.6.0.

The PyTorch code included with this repo doesn't use torch.fx and runs in PyTorch 1.6 okay - but if you are using other externally-sourced PyTorch code, it may use torch.fx module.

> This issue was resolved when I updated pytorch :) I have a question though, it seems my training is not using CUDA but I

Did you install the newer PyTorch from one of the pip wheels from this page?  These were built with CUDA support enabled.

https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-10-now-available/72048

FebrianFrancione, 2022-03-31:
Oh I see.
I did not know that.
I will have to try that tonight.
I got the training running but noticed half way thay my avg loss,
regression etc values were printing out as nan... worried the model
wouldn't be useful but just letting it run to see what happens.. since it
still says model is saved in detections.
Would you know why this is happening?

EG:
2022-03-30 20:20:18 - Epoch: 7, Step: 10/252, Avg Loss: nan, Avg Regression
Loss: nan, Avg Classification Loss: nan

JHPHELAN, 2022-04-10:
@FebrianFrancione 
I had this problem when I tried to run from my home directory instead of doing docker/run.sh first.  Worked fine when I launched the docker and went to the proper directory.

Movingforward1, 2022-04-28:
I am having the same torch.fx issue. Just over a month ago I was able to train models without this issue however now when I run train_ssd.py I receive the same error message. I tried updating to torch 1.8.0 but just received additional error messages about not having numpy.core._multiarray_umath

closed 2023-04-27 by dusty-nv