dusty-nv/jetson-inference #1396: NVIDIA server downloads fail

tensorturtle, 2022-03-31:
Hello @dusty-nv , I know that this is not your responsibility exactly, but several download links are (recently) broken for Jetson-related things, such as SD card images from https://developer.nvidia.com/embedded/downloads (404 Error) . Also,  SDK Manager returns the attached error. My co-worker and I have tried from two different countries. Please forward this problem to whomever is responsible for it in NVIDIA. Thank you.

[image]

dusty-nv, 2022-03-31:
Hi @tensorturtle, sorry about that, yes we are working to resolve the issue:

https://forums.developer.nvidia.com/t/tensorflow-download-removed/209866/2

tensorturtle, 2022-04-04:
Problem appears to have been fixed. Thank you.

closed 2022-04-04 by tensorturtle