dusty-nv/jetson-inference #1402: Randomness at inference time

aafaqin, 2022-04-19:
When I am running detection on JETSON NANO with 288x192 input size the detection time are different for all detection mostly it gives 10FPS but sometimes detection time drops to 5FPS and it happens randomly for 10% of the time. Any way I can reduce this randomness?
This is only detection time not preprocessing time.
Can someone please help me with this?

dusty-nv, 2022-04-19:
Hi @aafaqin, can you try running `sudo jetson_clocks.sh` script?

aafaqin, 2022-04-19:
Yes I have run that. Still the randomness Exists.

aafaqin, 2022-04-19:
I am doing RTSP but the FPS I am mentioning is only after the RTSP frame is acquired and then the inference function is called.

shithanshu-beltech, 2022-04-20:
@dusty-nv  can we run jetson_clocks.sh in production mode in a scenario when jetson board will be running 24X7

dusty-nv, 2022-04-20:
@shithanshu-beltech I would refer to the module datasheet to be sure, but with proper cooling yes it should be fine.  Also you can create your own nvpmodel profile with lower clocks if desired.  Or perhaps in your scenario you don't need jetson_clocks.sh

shithanshu-beltech, 2022-04-22:
Thanks @dusty-nv Will try it out and update you

closed 2023-04-27 by dusty-nv