dusty-nv/jetson-inference #1404: can not build jetson-inference from source error accured at depthnet.cpp

mrinmoy-sadhukhan, 2022-04-20:
[ 57%] Built target jetson-utils-python-27
/home/mrinmoy/jetson-inference/c/depthNet.h(190): error: identifier "COLORMAP_VIRIDIS_INVERTED" is undefined

/home/mrinmoy/jetson-inference/c/depthNet.h(180): error: identifier "COLORMAP_VIRIDIS_INVERTED" is undefined

2 errors detected in the compilation of "/tmp/tmpxft_0000703e_00000000-8_depthNet.compute_72.cpp1.ii".
CMake Error at jetson-inference_generated_depthNet.cu.o.cmake:279 (message):
  Error generating file
  /home/mrinmoy/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_depthNet.cu.o

CMakeFiles/jetson-inference.dir/build.make:75: recipe for target 'CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_depthNet.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_depthNet.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....[ 57%] Built target jetson-utils-python-27
/home/mrinmoy/jetson-inference/c/depthNet.h(190): error: identifier "COLORMAP_VIRIDIS_INVERTED" is undefined

/home/mrinmoy/jetson-inference/c/depthNet.h(180): error: identifier "COLORMAP_VIRIDIS_INVERTED" is undefined

2 errors detected in the compilation of "/tmp/tmpxft_0000703e_00000000-8_depthNet.compute_72.cpp1.ii".
CMake Error at jetson-inference_generated_depthNet.cu.o.cmake:279 (message):
  Error generating file
  /home/mrinmoy/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_depthNet.cu.o

CMakeFiles/jetson-inference.dir/build.make:75: recipe for target 'CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_depthNet.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_depthNet.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 58%] Building CXX object utils/display/gl-display-test/CMakeFiles/gl-display-test.dir/gl-display-test.cpp.o
[ 58%] Linking CXX executable ../../../aarch64/bin/gl-display-test
[ 58%] Built target gl-display-test
CMakeFiles/Makefile2:477: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

[ 58%] Building CXX object utils/display/gl-display-test/CMakeFiles/gl-display-test.dir/gl-display-test.cpp.o
[ 58%] Linking CXX executable ../../../aarch64/bin/gl-display-test
[ 58%] Built target gl-display-test
CMakeFiles/Makefile2:477: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2022-04-20:
Hi @mrinmoy-sadhukhan, are you on the latest master branch?   It seems like your `utils` submodule is missing or outdated

mrinmoy-sadhukhan, 2022-04-20:
sir today i once more download the repo and will rebuild it today I will send you report after the result comes.
thankyou

mrinmoy-sadhukhan, 2022-04-21:
i sucessfully build the repo thanks for suggestion It works..

closed 2022-04-21 by mrinmoy-sadhukhan