dusty-nv/jetson-inference #1405: build with python3.9

loupdmer, 2022-04-20:
Hi, I am trying to build for python3.9 version which is located in /usr/local directories.
I have adapted the cmake files to do this (with the latest version), however the shared objects for python use are located in /usr/lib/python3.9/dist-packages instead of /usr/local/lib/python3.9/site-packages.

I don't know how to change this in the cmake files, may somebody help me ?

dusty-nv, 2022-04-21:
Hi @loupdmer, you can change the Python install dirs here:

* https://github.com/dusty-nv/jetson-inference/blob/6cf3b12f503d64903ca1e77d4c5474d9b1513a4c/python/bindings/CMakeLists.txt#L72
* https://github.com/dusty-nv/jetson-utils/blob/def4a04d023960781a44f9cd97fd1464093becf0/python/bindings/CMakeLists.txt#L72

loupdmer, 2022-04-22:
Thank you very much for your help !

Have a nice day

closed 2022-04-22 by loupdmer