dusty-nv/jetson-inference #1406: [TRT] 2: [utils.cpp::checkMemLimit::380] Error Code 2: Internal Error (Assertion upperBound != 0 failed. Unknown embedded device

hgddingjun, 2022-04-22:
Hi@dusty-nv，I reference https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md build and install jetson-inference
but
when i try to run "./imagenet images/orange_0.jpg images/test/output_0.jpg" , the following error show up:
...
[TRT] Could not register plugin creator - ::FlattenConcat_TRT version 1
...
[TRT] 2: [utils.cpp::checkMemLimit::380] Error Code 2: Internal Error (Assertion upperBound != 0 failed. Unknown embedded device detected. Please update the table with the entry: {{1794, 6, 16}, 12653},)
[TRT] device GPU, failed to build CUDA engine
[TRT] device GPU, failed to load networks/bvlc_googlenet.caffemodel
[TRT] failed to load networks/bvlc_googlenet.caffemodel
[TRT] imageNet -- failed to initialize.
imagenet: failed to initialize imageNet
PS：I downloaded the models(bvlc_googlenet.caffemodel...) and put networks directory in advance.
Can you help me to solve the problem?

hgddingjun, 2022-04-22:
i use jetson xavier nv hardware platform.
jetpack sdk 4.6.1
Jetson_Linux_R32.7.1_aarch64.tbz2

leeyunhome, 2022-05-27:
> i use jetson xavier nv hardware platform. jetpack sdk 4.6.1 Jetson_Linux_R32.7.1_aarch64.tbz2

Hello, @hgddingjun 

hello? May I know which carrier board you are using?

Thank you.

ooijenvangilbert, 2022-06-11:
Hi, i am facing the same issue.

I am not running it from the docker container

My hardware is this board : https://www.seeedstudio.com/Jetson-20-1-H2-p-5329.html
jetpack 4.6.2 [L4T 32.7.2]
cuda 10.2.300
TensorRt 8.2.1.8
cuDNN 8.2.1.32

[TRT]    =============== Computing reformatting costs
[TRT]    *************** Autotuning Reformat: Float(307200,102400,320,1) -> Float(307200,1,960,3) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(input_0 -> <out>) (Reformat)
[TRT]    2: [utils.cpp::checkMemLimit::380] Error Code 2: Internal Error (Assertion upperBound != 0 failed. Unknown embedded device detected. Please update the table with the entry: {{1794, 6, 16}, 12660},)
[TRT]    device GPU, failed to build CUDA engine
[TRT]    device GPU, failed to load networks/FCN-ResNet18-Pascal-VOC-320x320/fcn_resnet18.onnx
[TRT]    segNet -- failed to load.
jetson.inference -- segNet failed to load network
Traceback (most recent call last):
  File "/home/xavier/Desktop/pypro/Nvidia/Deep-learning/segnet.py", line 33, in <module>
    net = jetson.inference.segNet('fcn-resnet18-voc')

ooijenvangilbert, 2022-06-15:
Hi @dusty-nv 

So i have a fresh docker container but inside and outside of the container running the same code i get this exception.

`[TRT] 2: [utils.cpp::checkMemLimit::380] Error Code 2: Internal Error (Assertion upperBound != 0 failed. Unknown embedded device detected. Please update the table with the entry: {{1794, 6, 16}, 12660},)`

Is there anything i can do to resolve this? do you need more info?

Thanks in advance

dusty-nv, 2022-06-15:
Hi @ooijenvangilbert, I believe you need to reflash your Jetson Xavier NX 16GB or SD card with the latest JetPack 5.0.1 - https://forums.developer.nvidia.com/t/i-have-stucked-when-i-use-jetson-inference-on-xavier-nx-16gb/214020/5?u=dusty_nv

ooijenvangilbert, 2022-06-16:
> Hi @ooijenvangilbert, I believe you need to reflash your Jetson Xavier NX 16GB or SD card with the latest JetPack 5.0.1 - https://forums.developer.nvidia.com/t/i-have-stucked-when-i-use-jetson-inference-on-xavier-nx-16gb/214020/5?u=dusty_nv

Hi @dusty-nv ,
I had the intention to do so but the SDK manager software (the pretty UI one) already crashed on startup and I don't feel secure enough for the command line option. Will however dive into it and if needed start a new item.

ooijenvangilbert, 2022-06-28:
> Hi @ooijenvangilbert, I believe you need to reflash your Jetson Xavier NX 16GB or SD card with the latest JetPack 5.0.1 - https://forums.developer.nvidia.com/t/i-have-stucked-when-i-use-jetson-inference-on-xavier-nx-16gb/214020/5?u=dusty_nv

After a very lengthy route of getting ubuntu 22.04 off my dual boot laptop and 18.04 installed i managed to get the Xavier flashed and up and running, this issue is resolved with JetPack 5.0.1

closed 2023-04-27 by dusty-nv