dusty-nv/jetson-inference #1407: Import torchvision = error

littlebee, 2022-04-25:
Hi!  👋  I set up my Jetson Nano as instructed here: https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md.  Verified using  JetPack 4.6.1.  I can also run the example `./imagenet.py images/orange_0.jpg ~/inference_test.jpg`.  Works great!  but...

When I try to `import torchvision` from the python3 console, this happens:
```
bee@scatbot-nano:~/jetson-inference/build/aarch64/bin$ python3
Python 3.6.9 (default, Mar 15 2022, 13:55:28)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torchvision
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0-py3.6-linux-aarch64.egg/torchvision/__init__.py", line 6, in <module>
    from torchvision import datasets
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0-py3.6-linux-aarch64.egg/torchvision/datasets/__init__.py", line 9, in <module>
    from .fakedata import FakeData
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0-py3.6-linux-aarch64.egg/torchvision/datasets/fakedata.py", line 3, in <module>
    from .. import transforms
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0-py3.6-linux-aarch64.egg/torchvision/transforms/__init__.py", line 1, in <module>
    from .transforms import *
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0-py3.6-linux-aarch64.egg/torchvision/transforms/transforms.py", line 15, in <module>
    from . import functional as F
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0-py3.6-linux-aarch64.egg/torchvision/transforms/functional.py", line 302, in <module>
    def resize(img, size, interpolation=Image.BILINEAR):
AttributeError: module 'PIL.Image' has no attribute 'BILINEAR'
```
Looks like an old version of torchvision is maybe installed, but IDK? 

Any help you can provide would be very appreciated.   Thank you!

dusty-nv, 2022-04-25:
> AttributeError: module 'PIL.Image' has no attribute 'BILINEAR'

Hi @littlebee, can you try running `pip3 install 'pillow<7'`

littlebee, 2022-04-26:
@dusty-nv,  `pip install 'pillow'` got me passed that error.  Now on to the next...

```
terminate called after throwing an instance of 'c10::Error'
  what():  quantized engine QNNPACK is not supported
Exception raised from setQEngine at ../aten/src/ATen/Context.cpp:132 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) + 0xa0 (0x7f490109a8 in /home/bee/.local/lib/python3.6/site-packages/torch/lib/libc10.so)
frame #1: at::Context::setQEngine(c10::QEngine) + 0x1b4 (0x7f7a7f5b94 in /home/bee/.local/lib/python3.6/site-packages/torch/lib/libtorch_cpu.so)
frame #2: THPModule_setQEngine(_object*, _object*) + 0x94 (0x7f7dd5a6bc in /home/bee/.local/lib/python3.6/site-packages/torch/lib/libtorch_python.so)
```

I'm trying to run the benchmark code from [this article](https://pytorch.org/tutorials/intermediate/realtime_rpi.html) to see how the nano compares to the PI4:

```python
#!/usr/bin/env python3

import time

import torch
import numpy as np
from torchvision import models, transforms

import cv2
from PIL import Image

torch.backends.quantized.engine = 'qnnpack'

cap = cv2.VideoCapture(0, cv2.CAP_V4L2)
cap.set(cv2.CAP_PROP_FRAME_WIDTH, 224)
cap.set(cv2.CAP_PROP_FRAME_HEIGHT, 224)
cap.set(cv2.CAP_PROP_FPS, 36)

preprocess = transforms.Compose([
    transforms.ToTensor(),
    transforms.Normalize(mean=[0.485, 0.456, 0.406],
                         std=[0.229, 0.224, 0.225]),
])

net = models.quantization.mobilenet_v2(pretrained=True, quantize=True)
# jit model to take it from ~20fps to ~30fps
net = torch.jit.script(net)

started = time.time()
last_logged = time.time()
frame_count = 0

with torch.no_grad():
    while True:
        # read frame
        ret, image = cap.read()
        if not ret:
            raise RuntimeError(f"failed to read frame. ret={ret}")

        # convert opencv output from BGR to RGB
        image = image[:, :, [2, 1, 0]]
        permuted = image

        # preprocess
        input_tensor = preprocess(image)

        # create a mini-batch as expected by the model
        input_batch = input_tensor.unsqueeze(0)

        # run model
        output = net(input_batch)
        # do something with output ...

        # log model performance
        frame_count += 1
        now = time.time()
        if now - last_logged > 1:
            print(f"{frame_count / (now-last_logged)} fps")
            last_logged = now
            frame_count = 0
```

littlebee, 2022-04-26:
BTW, if you are curious, I was able to get the above to run by removing `torch.backends.quantized.engine = 'qnnpack'` and removing the `quantize=True` from the init of mobilenet.  The result was not so good.   In fairness, the Pi4 runs with the two changes caused a drop from 30fps to 2-3fps.  Jetson Nano with this build was < 1fps with the changes.  😨 
```
bee@scatbot-nano:~/scatbot$ python3.6 debug/benchmark-pytorch.py
Downloading: "https://download.pytorch.org/models/mobilenet_v2-b0353104.pth" to /home/bee/.cache/torch/hub/checkpoints/mobilenet_v2-b0353104.pth
100.0%
0.10070409485417284 fps
0.1815559611964164 fps
0.18365771485585142 fps
0.1833795895296882 fps
0.1842552758359428 fps
0.18287348679233767 fps
0.18260259495441253 fps
0.18218296359184905 fps
0.18333008646092672 fps
0.182871118733919 fps
0.18393394259265 fps
0.1851224670260713 fps
0.1823429043232569 fps
0.18341102374283458 fps
0.1825273894178081 fps
0.18326822117306754 fps
0.18252977241266372 fps
0.1834141276539771 fps
```

dusty-nv, 2022-04-26:
> Jetson Nano with this build was < 1fps with the changes. 😨

I don't think this is using the GPU.  you need to call `.cuda()` on the model and tensors in order for PyTorch to use the GPU

littlebee, 2022-04-27:
Gtk!  I will try that soon and leave an update here.  Thanks for replying!

closed 2023-04-27 by dusty-nv