dusty-nv/jetson-inference #1409: Open CV Version Error

zdrescher, 2022-05-02:
We where previously leveraging `dustynv/jetson-inference:r32.6.1` image for an openCV based inference application.

We are getting an issue with the cuDNN and OpenCV having incompatible versions. 

```
[ WARN:0] global /opt/opencv/modules/dnn/src/cuda4dnn/init.hpp (34) checkVersions cuDNN reports version 8201 which does not match with the version 8000 with which OpenCV was built
```

We are attempting to test on a device with L4T version R32.7.1, I upgraded the base image to leverage `dustynv/jetson-inference:r32.7.1` but still getting the same issue.

dusty-nv, 2022-05-02:
Hmmm unfortunately it appears that OpenCV would need rebuilt every time cuDNN changes - I was not previously aware of this

For now, you may wish to run the `scripts/docker_build_opencv.sh` script from jetson-containers to rebuild OpenCV and give you the deb files which you can then install into the container by changing the URL here and rebuilding it:   https://github.com/dusty-nv/jetson-inference/blob/6cf3b12f503d64903ca1e77d4c5474d9b1513a4c/docker/build.sh#L62

zdrescher, 2022-05-03:
Sounds good,

While I have you, do you have any thoughts on the best image to use for an openCV based pipeline (we are using the [DNN module](https://docs.opencv.org/4.x/d3/df1/classcv_1_1dnn_1_1DetectionModel.html))? Ultimately, this `jetson-inference` image was the first I was able to get it to run on but i noticed [l4t-ml](https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-ml) has openCV as well. I dont recall if i tried that one initially. I just tried `l4t-ml:r32.7.1-py3` but its getting the same cuDNN incompatibility.

I am going suggest my team roll back to r32.6.1 for now but please let me know if there are any fixes to the other versions.

Cheers,

dusty-nv, 2022-05-03:
> I just tried `l4t-ml:r32.7.1-py3` but its getting the same cuDNN incompatibility.

Yea, those are using the same OpenCV packages that I built previously.  I don't use OpenCV's DNN module so was unaware of the issue.  

> I am going suggest my team roll back to r32.6.1 for now but please let me know if there are any fixes to the other versions.
That would seem like a fine approach unless you need other stuff in r32.7.1.  Otherwise you would want to rebuild OpenCV on r32.7.1 and then install it in the containers, overwriting my previous version of OpenCV.

zdrescher, 2022-05-05:
Attempting to build openCV in R32.7.1 using `scripts/docker_build_open.cv` from jetson-containers is also failing during the Cmake step:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_LIBRARY (ADVANCED)
    linked by target "opencv_cudev" in directory /opt/opencv_contrib/modules/cudev
    linked by target "opencv_core" in directory /opt/opencv/modules/core
    linked by target "opencv_cudaarithm" in directory /opt/opencv_contrib/modules/cudaarithm
```

With many more similar lines following. I can open a similar issue in Jetson containers, if that is a better place to track.

dusty-nv, 2022-05-05:
Hi @zdrescher, can you make sure that you have cublas libraries installed on your device (outside of container)?

```bash
$ ls /usr/lib/aarch64-linux-gnu/libcublas*
/usr/lib/aarch64-linux-gnu/libcublasLt.so            /usr/lib/aarch64-linux-gnu/libcublas.so
/usr/lib/aarch64-linux-gnu/libcublasLt.so.10         /usr/lib/aarch64-linux-gnu/libcublas.so.10
/usr/lib/aarch64-linux-gnu/libcublasLt.so.10.2.2.89  /usr/lib/aarch64-linux-gnu/libcublas.so.10.2.2.89
/usr/lib/aarch64-linux-gnu/libcublasLt_static.a      /usr/lib/aarch64-linux-gnu/libcublas_static.a
```

Also do you have your default runtime set to `nvidia`?   https://github.com/dusty-nv/jetson-containers#docker-default-runtime

closed 2023-04-27 by dusty-nv