dusty-nv/jetson-inference #1420: What's the unit of the Raw Depth Field in depthnet?

Hazeline2018, 2022-05-24:
Hi Dusty, I'm working on mono-depth algorithm and testing with provided "depthnet" model. I can get the Raw Depth Field readings by calling GetDepthField() method. But I'm confused about the real meaning of the FP numbers. As far as I understand, the depth number for each pixel (depending on the --depth-size scale) is relative depth estimate. But what is the unit of the depth? If it's a relative ratio, then reference to what? For example, on one image, the range of depth field is (min=0.8974, max=4.3873), are they normalized?

thanks!
Terry

Hazeline2018, 2022-05-24:
Better yet, is there detailed explanation doc about depthnet model other than just the depthnet.md? I searched API ref doc, but no much details.

dusty-nv, 2023-04-27:
@Hazeline2018 sorry for the delay - during visualization (rendering of the colorized image), histogram equalization is used, but there is no post-processing applied to the raw depth field, and I'm not sure what normalization the model itself applies.   The models came from https://fastdepth.mit.edu/

closed 2023-04-27 by dusty-nv