dusty-nv/jetson-inference #1421: Posenet only processes 1 frame of the video

gilles-syrto, 2022-05-24:
Hi ! 

As much as I really appreciates the posenet application, I can not get why for some input videos it works absolutely fine, and for some others, it only processes 1 frame and not the whole video. I can reproduce this behaviour with both the python example and the executable that I have in my _.../build/aatch64/bin_ folder... 
At first I thought it might be related to the frame rate of the video, but I tried with different frame rates and did not get any better results. Also, I could not figure it out by reading the logs...

My setup :
- Jetson Nano 2GB with JP 4.6
- cloned this repo and built it from sources

Here is the log I get with the executable : 

`
./posenet ../../../data/input/dataset1.mp4 ../../../data/output/test-bin.mp4
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for ../../../data/input/dataset1.mp4
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- discovered video resolution: 320x240  (framerate 10.204082 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)2, profile=(string)high, width=(int)320, height=(int)240, framerate=(fraction)500/49, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=../../../data/input/dataset1.mp4 ! qtdemux ! queue ! h264parse ! omxh264dec ! video/x-raw(memory:NVMM) ! appsink name=mysink
[video]  created gstDecoder from file:///home/syrto/work/jetson-inference/build/aarch64/bin/../../../data/input/dataset1.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///home/syrto/work/jetson-inference/build/aarch64/bin/../../../data/input/dataset1.mp4
     - protocol:  file
     - location:  ../../../data/input/dataset1.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      h264
  -- width:      320
  -- height:     240
  -- frameRate:  10.204082
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  h264parse ! qtmux ! filesink location=../../../data/output/test-bin.mp4
[video]  created gstEncoder from file:///home/syrto/work/jetson-inference/build/aarch64/bin/../../../data/output/test-bin.mp4
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: file:///home/syrto/work/jetson-inference/build/aarch64/bin/../../../data/output/test-bin.mp4
     - protocol:  file
     - location:  ../../../data/output/test-bin.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     output
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  30.000000
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[OpenGL] failed to create X11 Window.
[... 346 lines cut ...]
[TRT]    Pre-Process   CPU   0.07724ms  CUDA   1.07526ms
[TRT]    Network       CPU 5429.70264ms  CUDA 5428.16162ms
[TRT]    Post-Process  CPU  27.62575ms  CUDA  27.62161ms
[TRT]    Visualize     CPU  15.07111ms  CUDA  15.07135ms
[TRT]    Total         CPU 5472.47705ms  CUDA 5471.92969ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[gstreamer] gstDecoder -- end of stream (EOS) has been reached, stream has been closed
posenet: shutting down...
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL
H264: Profile = 66, Level = 40
NVMEDIA_ENC: bBlitMode is set to TRUE
[gstreamer] gstDecoder -- pipeline stopped
[gstreamer] gstEncoder -- shutting down pipeline, sending EOS
[gstreamer] gstEncoder -- transitioning pipeline to GST_STATE_NULL
[gstreamer] gstEncoder -- pipeline stopped
posenet: shutdown complete.

`

gilles-syrto, 2022-05-24:
Also, I did try the : 

`
cmake -DENABLE_NVMM=OFF ../
make
sudo make install
`

which removed quite a lot of log lines, but did not really help with the real problem.

dusty-nv, 2022-05-24:
> I can not get why for some input videos it works absolutely fine, and for some others, it only processes 1 frame and not the whole video.

Hi @gilles-syrto, can you try playing the video with `video-viewer` or `video-viewer.py` and seeing if it's related to posenet at all?

If video-viewer can't play it either, I think what I would recommend is transcoding the video to a different codec or re-encoding it, because perhaps the video has something different about it where GStreamer is unable to play it correctly.

gilles-syrto, 2022-05-31:
> Hi @gilles-syrto, can you try playing the video with `video-viewer` or `video-viewer.py` and seeing if it's related to posenet at all?
>
> If video-viewer can't play it either, I think what I would [...]

Thanks for your reply. I was offline the last few days, but I'll try it in the next days and let you know if it is related to the encoding of the video or not. 

Gilles

gilles-syrto, 2022-06-01:
Hi ! video-viewer is able to play the video that posenet is not. Any idea what I could check ?