dusty-nv/jetson-inference #1432: About the problems after make reaches 50%

JJXuanYuan, 2022-06-01:
../../aarch64/lib/libjetson-inference.so：对‘trt_pose::parse::connect_parts_out_workspace(int, int)’未定义的引用
../../aarch64/lib/libjetson-inference.so：对‘trt_pose::parse::assignment_out_workspace(int)’未定义的引用
../../aarch64/lib/libjetson-inference.so：对‘trt_pose::parse::connect_parts_out_batch(int*, int*, int const*, int const*, int const*, int, int, int, int, int, void*)’未定义的引用
../../aarch64/lib/libjetson-inference.so：对‘trt_pose::parse::assignment_out_nk(int*, float const*, int const*, int const*, int, int, int, int, float, void*)’未定义的引用
../../aarch64/lib/libjetson-inference.so：对‘trt_pose::parse::paf_score_graph_out_nkhw(float*, int const*, float const*, int const*, float const*, int, int, int, int, int, int, int)’未定义的引用
../../aarch64/lib/libjetson-inference.so：对‘trt_pose::parse::refine_peaks_out_nchw(float*, int const*, int const*, float const*, int, int, int, int, int, int)’未定义的引用
../../aarch64/lib/libjetson-inference.so：对‘trt_pose::parse::find_peaks_out_nchw(int*, int*, float const*, int, int, int, int, int, float, int)’未定义的引用
collect2: error: ld returned 1 exit status
examples/imagenet/CMakeFiles/imagenet.dir/build.make:101: recipe for target 'aarch64/bin/imagenet' failed
make[2]: *** [aarch64/bin/imagenet] Error 1
CMakeFiles/Makefile2:175: recipe for target 'examples/imagenet/CMakeFiles/imagenet.dir/all' failed
make[1]: *** [examples/imagenet/CMakeFiles/imagenet.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

JJXuanYuan, 2022-06-01:
问题解决了，没有什么特殊问题，如果有问题就把build重新建一个然后重新cmake应该全是网络问题，值得一提的是pulgins下面路径有一个poss一般都下不下来，来到这里单独下载一份然后放到对应路径就行了，以上结论让我浪费了儿童节的整整一个下午，我真得会谢

dusty-nv, 2022-06-01:
Hi @JJXuanYuan, undo your changes to the code and run the program with the `--input-flip=rotate-180` option:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options

If the camera still isn't starting, you may need to reboot your system first.

closed 2023-04-27 by dusty-nv