dusty-nv/jetson-inference #1433: 2 Questions about re-training detection ssd on your own dataset

r2ba7, 2022-06-03:
First question, can I train custom dataset using Mobilenet V2 SSD or only V1 available ?
Second question, I had a dataset which I took photos of, and I've a csv file which includes the columns as attached to the picture, should I do this from all over again like u suggested in [https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md](url)
[image]

dusty-nv, 2022-06-03:
Hi @r2ba7, only the SSD-Mobilenet-v1 is tested + working with the ONNX export and import into TensorRT.   SSD-Mobilenet-v2 doesn't offer much improvement anyways.

You can either write a tool or script to convert your CSV to a dataset in the Pascal VOC format, or use CVAT.org to re-label your images and export as Pascal VOC from cvat.

r2ba7, 2022-06-03:
@dusty-nv I want to try inception and mobilenet for research purposes also, thats why I was asking about mobilenet v2 or inception compatibility if that's possible

dusty-nv, 2022-06-03:
I believe these are the different network architectures that train_ssd.py supports:   https://github.com/qfgaohao/pytorch-ssd/blob/f61ab424d09bf3d4bb3925693579ac0a92541b0d/run_ssd_example.py#L22

However the only one that I have tested + working with ONNX is SSD-Mobilenet-v1.   
If you are conducting research and aren't as concerned with deployment, then you can use other object detection code with PyTorch/DarkNet/TensorFlow/ect and not be concerned with the TensorRT aspect.

closed 2023-04-27 by dusty-nv