dusty-nv/jetson-inference #1437: (video-viewer and detectnet) EGL display error and execute: 751 Failed to create CaptureSession

emma-jinger, 2022-06-09:
### Device and Setup
Device: Jetson Nano Developer Kit (flashed in headless mode)
Jetpack version: 
` Version: 4.6.2-b5
   Installed-Size: 199 kB
   Depends: nvidia-cuda (= 4.6.2-b5), nvidia-opencv (= 4.6.2-b5), nvidia-cudnn8 (= 4.6.2-b5), nvidia-tensorrt (= 4.6.2-b5), nvidia-visionworks (= 4.6.2-b5), nvidia-container (= 4.6.2-b5), nvidia-vpi (= 4.6.2-b5), nvidia-l4t-jetson-multimedia-api (>> 32.7-0), nvidia-l4t-jetson-multimedia-api (<< 32.8-0)`
L4T: R32 (release), REVISION: 7.1, GCID: 29818004, BOARD: t210ref
Csi camera connected to the Nano device: https://www.amazon.com/dp/B07T43K7LC/ref=redir_mobile_desktop?_encoding=UTF8&psc=1&ref_=ya_aw_od_pi

The Nano device is sshed into from my Ubuntu PC by `ssh -X Nano@NanoIP`.

I followed this link to build jetson-inference from Source: [https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md)

### Issues
After the build, running the video-viewer with `video-viewer csi://0  rtp://192.168.1.159:1234` produced error as follows:
`nvbuf_utils: Could not get EGL display connection
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: rotate-180
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  rtph264pay config-interval=1 ! udpsink host=192.168.1.159 port=1234 auto-multicast=true
[video]  created gstEncoder from rtp://192.168.1.159:1234
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtp://192.168.1.159:1234
     - protocol:  rtp
     - location:  192.168.1.159
     - port:      1234
  -- deviceType: ip
  -- ioType:     output
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  30.000000
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1366x768
[OpenGL] glDisplay -- X window resolution:    1366x768
[OpenGL] glDisplay -- display device initialized (1366x768)
[video]  created glDisplay from display://0
------------------------------------------------
[... 268 lines cut ...]
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer message stream-start ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:751 Failed to create CaptureSession
[gstreamer] gstCamera -- end of stream (EOS)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[gstreamer] gstreamer pipeline0 recieved EOS signal...
[gstreamer] gstDecoder -- failed to retrieve next image buffer
detectnet:  failed to capture video frame
[gstreamer] gstDecoder -- failed to retrieve next image buffer
detectnet:  failed to capture video frame
^Z
[2]+  Stopped                 detectnet csi://0 rtp://192.168.1.159:1234
`
Any insights would be appreciated. I've been spending hours trying to understand and solve this issue, but in vain.

dusty-nv, 2022-06-10:
> The Nano device is sshed into from my Ubuntu PC by `ssh -X Nano@NanoIP`
> `NvEGLImageFromFd: No EGLDisplay to create EGLImage`

If you have X11 tunneling enabled over SSH, can you disable it?

Otherwise, try building jetson-inference with the `cmake -DENABLE_NVMM=off ../` option

emma-jinger, 2022-06-10:
Thank you for such quick response, Dustin! 

I changed `ForwardX11` to `no `  in `/etc/ssh/ssh_config` in the Nano device, and then sshed into the Nano device without the option -X. I reran `video-viewer` using video-viewer `csi://0 rtp://192.168.1.159:1234`,  the same errors (EGL display connection and Execute: 751 failed to create CaptureSession) occurred.  

As for your second suggestion, could you kindly explain what ` -DENABLE_NVMM ` option for `cmake` does? And how to properly purge the current jetson-inference from the Nano device before I rebuild it? 

I appreciate your guidance! Thank you :)

desegura, 2022-06-27:
Same error here @dusty-nv tried `git pull` and `cmake -DENABLE_NVMM=off ../` but not working 

```
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=(fraction)60/1
[gstreamer] gstBufferManager -- recieved first frame, codec=raw format=nv12 width=1920 height=1080 size=1008
[gstreamer] gstBufferManager -- recieved NVMM memory
NvEGLImageFromFd: No EGLDisplay to create EGLImage
[gstreamer] gstBufferManager -- failed to map EGLImage from NVMM buffer
[gstreamer] gstCamera -- failed to handle incoming buffer
[gstreamer] gstBufferManager -- map buffer size was less than max size (1008 vs 1015)
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=(fraction)60/1
[gstreamer] gstBufferManager -- recieved first frame, codec=raw format=nv12 width=1920 height=1080 size=1015
[gstreamer] gstBufferManager -- recieved NVMM memory
NvEGLImageFromFd: No EGLDisplay to create EGLImage
[gstreamer] gstBufferManager -- failed to map EGLImage from NVMM buffer
[gstreamer] gstCamera -- failed to handle incoming buffer
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
```

dusty-nv, 2022-06-27:
> tried `git pull` and `cmake -DENABLE_NVMM=off ../` but not working

Did you also run `make && sudo make install` after running that cmake command?

desegura, 2022-06-27:
yes 🤣

dusty-nv, 2022-06-27:
OK, if it was in fact built with `-DENABLE_NVMM=off`, then you would not be recieving messages like this:

``` bash
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=(fraction)60/1
[gstreamer] gstBufferManager -- recieved NVMM memory
NvEGLImageFromFd: No EGLDisplay to create EGLImage
[gstreamer] gstBufferManager -- failed to map EGLImage from NVMM buffer
```

desegura, 2022-06-27:
yes yes I received this message once enabled NVMM back again
now I'm trying with a fresh SD install

Is there any way to run any checkup to discard hardware malfunction due to overheating?

Thanks

emma-jinger, 2022-07-05:
I have re-built jetson-inference. I followed [this link](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) and build it with cmake --DENABLE_NVMM=off. 

I got the error like below after running `video-viewer csi://0 rtp://192.168.1.159:1234`. A blank display window popped up. It didn't show anything.
```
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
RingBuffer -- allocated 4 buffers (230400 bytes each, 921600 bytes total)
video-viewer:  captured 1 frames (320 x 240)
[OpenGL] glDisplay -- set the window size to 320x240
[OpenGL] creating 320x240 texture (GL_RGB8 format, 230400 bytes)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/display/glTexture.cpp:360
RingBuffer -- allocated 2 buffers (115200 bytes each, 230400 bytes total)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/cuda/cudaYUV-YV12.cu:257
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/cuda/cudaColorspace.cpp:128
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/codec/gstEncoder.cpp:562
[gstreamer] gstEncoder::Render() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f
video-viewer:  captured 2 frames (320 x 240)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/display/glTexture.cpp:360
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/cuda/cudaYUV-YV12.cu:257
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/cuda/cudaColorspace.cpp:128
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/jetson-inference/utils/codec/gstEncoder.cpp:562
[gstreamer] gstEncoder::Render() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f

```
Runnining `detectnet csi://0 rtp://192.168.1.159:1234` produced the same error.
 
[image]

I saw someone else have the same issue from [this link](url). So I tried all the commands Dustin suggested there, still not working. 
The output of the command ```gst-launch-1.0 -v videotestsrc ! video/x-raw, width=300,height=300,framerate=30/1 ! x264enc ! rtph264pay ! udpsink host=192.168.1.159 port=5000``` is:
```
emma_dev22@Jetson-Nano:/home/jetson-inference/build/aarch64/bin$ gst-launch-1.0 -v videotestsrc ! video/x-raw, width=300,height=300,framerate=30/1 ! x264enc ! rtph264pay ! udpsink host=192.168.1.159 port=5000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
Redistribute latency...
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:src: caps = video/x-h264, codec_data=(buffer)01640014ffe1001d67640014acd941309fbbc05a830082d280000003008000001e478a14cb01000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)2, profile=(string)high, width=(int)300, height=(int)300, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)640014, sprop-parameter-sets=(string)"Z2QAFKzZQTCfu8BagwCC0oAAAAMAgAAAHkeKFMs\=\,aOvssiw\=", payload=(int)96, ssrc=(uint)3030945000, timestamp-offset=(uint)1495392018, seqnum-offset=(uint)4020, a-framerate=(string)30
/GstPipeline:pipeline0/GstUDPSink:udpsink0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)640014, sprop-parameter-sets=(string)"Z2QAFKzZQTCfu8BagwCC0oAAAAMAgAAAHkeKFMs\=\,aOvssiw\=", payload=(int)96, ssrc=(uint)3030945000, timest [... cut ...] 601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: timestamp = 1495392018
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: seqnum = 4020
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```
Wondering if @dusty-nv happened to have found a solution for that?

dusty-nv, 2022-07-05:
Hi @emma-jinger, are you able to view your camera with `nvgstcapture-1.0` program?

Do you have an actual display attached to your Jetson, or are you running over SSH?

emma-jinger, 2022-07-05:
Hi @dusty-nv, I'm running over SSH.  
To be honest, I'm not familiar with `nvgstcapture-1.0` program. Below is my attempt to use it after a quick read from https://docs.nvidia.com/jetson/archives/r34.1/DeveloperGuide/text/SD/Multimedia/AcceleratedGstreamer.html#nvgstcapture-1-0-reference
```
emma_dev22@Jetson-Nano:/home/jetson-inference/build/aarch64/bin$ nvgstcapture-1.0 --camsrc 1 -v 0
Encoder null, cannot set bitrate!
Encoder Profile = High
Supported resolutions in case of ARGUS Camera
  (2) : 640x480
  (3) : 1280x720
  (4) : 1920x1080
  (5) : 2104x1560
  (6) : 2592x1944
  (7) : 2616x1472
  (8) : 3840x2160
  (9) : 3896x2192
  (10): 4208x3120
  (11): 5632x3168
  (12): 5632x4224

Runtime ARGUS Camera Commands:

  Help : 'h'
  Quit : 'q'
  Set Capture Mode:
      mo:<val>
          (1): image
          (2): video
  Get Capture Mode:
      gmo
  Set sensor orientation:
      so:<val>
          (0): none
          (1): Rotate counter-clockwise 90 degrees
          (2): Rotate 180 degrees
          (3): Rotate clockwise 90 degrees
  Get sensor orientation:
      gso
  Set sensor mode:
      smo:<val> e.g., smo:1
  Get sensor mode:
      gsmo
  Set Whitebalance Mode:
      wb:<val>
          (0): off
          (1): auto
          (2): incandescent
          (3): fluorescent
          (4): warm-fluorescent
          (5): daylight
          (6): cloudy-daylight
          (7): twilight
          (8): shade
          (9): manual
  Get Whitebalance Mode:
      gwb
  Set Saturation (0 to 2):
      st:<val> e.g., st:1.25
  Get Saturation:
      gst
  Set Exposure Compensation (-2 to 2):
      ec:<val> e.g., ec:-2
  Get Exposure Compensation:
      gec
  Set Auto Whitebalance Lock:
      awbl:<val> e.g., awbl:0
  Get Auto Whitebalance Lock:
      awbl
  Set Auto Exposure Lock:
      ael:<val> e.g., ael:0
  Get Auto Exposure Lock:
      gael
  Set TNR Mode:
      tnrm:<val> e.g., tnrm:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get TNR Mode:
      gtnrm
  Set TNR Strength (-1 to 1):
      tnrs:<val> e.g., tnrs:0.5
  Get TNR Strength:
      gtnrs
  Set EE Mode:
      eem:<val> e.g., eem:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get EE Mode:
      geem
  Set EE Strength (-1 to 1):
      ees:<val> e.g., ees:0.5
  Get EE Strength:
      gees
  Set Auto Exposure Anti-Banding (0 to 3):
      aeab:<val> e.g., aeab:2
          (0): OFF
          (1): MODE AUTO
          (2): MODE 50HZ
          (3): MODE 60HZ
  Get Auto Exposure Anti-Banding:
      gaeab
  Set Gain Range:
      gr:<val><space><val> e.g., gr:1 16
  Get Gain Range:
      ggr
  Set Exposure Time Range:
      etr:<val><space><val> e.g., etr:34000 35000
  Get Exposure Time Range:
      getr
  Set ISP Digital Gain Range:
      dgr:<val><space><val> e.g., dgr:2 152
  Get ISP Digital Gain Range:
      gdgr
  Capture: enter 'j' OR
           followed by a timer (e.g., jx5000, capture after 5 seconds) OR
           followed by multishot count (e.g., j:6, capture 6 images)
           timer/multihot values are optional, capture defaults to single shot with timer=0s
  Start Recording : enter '1'
  Stop Recording  : enter '0'
  Video snapshot  : enter '2' (While recording video)
  Get Preview Resolution:
      gpcr
  Get Image Capture Resolution:
      gicr
  Get Video Capture Resolution:
      gvcr

Runtime encoder configuration options:

  Set Encoding Bit-rate(in bytes):
      br:<val> e.g., br:4000000
  Get Encoding Bit-rate(in bytes):
      gbr
  Set Encoding Profile(only for H.264):
      ep:<val> e.g., ep:1
          (0): Baseline
          (1): Main
          (2): High
  Get Encoding Profile(only for H.264):
      gep
  Force IDR Frame on video Encoder(only for H.264):
      Enter 'f' 

nvbuf_utils: Could not get EGL display connection
bitrate = 4000000
Encoder Profile = High
Encoder control-rate = 1
Encoder EnableTwopassCBR = 0
Opening in BLOCKING MODE 
** Message: 13:35:33.029: <main:4674> iterating capture loop ....
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
```

dusty-nv, 2022-07-06:
For MIPI CSI camera, you should just be able to run it without additional arguments.

Regardless, does it work when you ran it before?

emma-jinger, 2022-07-06:
With or without the additional arguments, the output in the terminal is the same as I posted above and no display window is shown:(

dusty-nv, 2022-07-06:
OK, if nvgstcapture-1.0 is unable to display it, then neither will jetson-inference's video-viewer tool.  I suspect their is something wrong with your camera or it's driver.  Which camera are you using?

emma-jinger, 2022-07-06:
Got it (IMX219 Camera) from here https://www.amazon.com/dp/B07T43K7LC/ref=redir_mobile_desktop?_encoding=UTF8&psc=1&ref_=ya_aw_od_pi 
The camera works fine with `video-viewer` or `detectnet` program when I attach a monitor to the Nano device.

emma-jinger, 2022-07-06:
I just changed `XForwarding` in the nano device `sshd_config `file to `no`  and `ForwardX11` to no in my local pc. 
After I sshed into the nano again without -X option from my local pc (Ubuntu 20.04), I ran the command  nvgstcapture-1.0 again, and the display window still is not shown. The output is 
```
emma_dev22@Jetson-Nano:~$ nvgstcapture-1.0
Encoder null, cannot set bitrate!
Encoder Profile = High
Supported resolutions in case of ARGUS Camera
  (2) : 640x480
  (3) : 1280x720
  (4) : 1920x1080
  (5) : 2104x1560
  (6) : 2592x1944
  (7) : 2616x1472
  (8) : 3840x2160
  (9) : 3896x2192
  (10): 4208x3120
  (11): 5632x3168
  (12): 5632x4224

Runtime ARGUS Camera Commands:

  Help : 'h'
  Quit : 'q'
  Set Capture Mode:
      mo:<val>
          (1): image
          (2): video
  Get Capture Mode:
      gmo
  Set sensor orientation:
      so:<val>
          (0): none
          (1): Rotate counter-clockwise 90 degrees
          (2): Rotate 180 degrees
          (3): Rotate clockwise 90 degrees
  Get sensor orientation:
      gso
  Set sensor mode:
      smo:<val> e.g., smo:1
  Get sensor mode:
      gsmo
  Set Whitebalance Mode:
      wb:<val>
          (0): off
          (1): auto
          (2): incandescent
          (3): fluorescent
          (4): warm-fluorescent
          (5): daylight
          (6): cloudy-daylight
          (7): twilight
          (8): shade
          (9): manual
  Get Whitebalance Mode:
      gwb
  Set Saturation (0 to 2):
      st:<val> e.g., st:1.25
  Get Saturation:
      gst
  Set Exposure Compensation (-2 to 2):
      ec:<val> e.g., ec:-2
  Get Exposure Compensation:
      gec
  Set Auto Whitebalance Lock:
      awbl:<val> e.g., awbl:0
  Get Auto Whitebalance Lock:
      awbl
  Set Auto Exposure Lock:
      ael:<val> e.g., ael:0
  Get Auto Exposure Lock:
      gael
  Set TNR Mode:
      tnrm:<val> e.g., tnrm:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get TNR Mode:
      gtnrm
  Set TNR Strength (-1 to 1):
      tnrs:<val> e.g., tnrs:0.5
  Get TNR Strength:
      gtnrs
  Set EE Mode:
      eem:<val> e.g., eem:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get EE Mode:
      geem
  Set EE Strength (-1 to 1):
      ees:<val> e.g., ees:0.5
  Get EE Strength:
      gees
  Set Auto Exposure Anti-Banding (0 to 3):
      aeab:<val> e.g., aeab:2
          (0): OFF
          (1): MODE AUTO
          (2): MODE 50HZ
          (3): MODE 60HZ
  Get Auto Exposure Anti-Banding:
      gaeab
  Set Gain Range:
      gr:<val><space><val> e.g., gr:1 16
  Get Gain Range:
      ggr
  Set Exposure Time Range:
      etr:<val><space><val> e.g., etr:34000 35000
  Get Exposure Time Range:
      getr
  Set ISP Digital Gain Range:
      dgr:<val><space><val> e.g., dgr:2 152
  Get ISP Digital Gain Range:
      gdgr
  Capture: enter 'j' OR
           followed by a timer (e.g., jx5000, capture after 5 seconds) OR
           followed by multishot count (e.g., j:6, capture 6 images)
           timer/multihot values are optional, capture defaults to single shot with timer=0s
  Start Recording : enter '1'
  Stop Recording  : enter '0'
  Video snapshot  : enter '2' (While recording video)
  Get Preview Resolution:
      gpcr
  Get Image Capture Resolution:
      gicr
  Get Video Capture Resolution:
      gvcr

Runtime encoder configuration options:

  Set Encoding Bit-rate(in bytes):
      br:<val> e.g., br:4000000
  Get Encoding Bit-rate(in bytes):
      gbr
  Set Encoding Profile(only for H.264):
      ep:<val> e.g., ep:1
          (0): Baseline
          (1): Main
          (2): High
  Get Encoding Profile(only for H.264):
      gep
  Force IDR Frame on video Encoder(only for H.264):
      Enter 'f' 

bitrate = 4000000
Encoder Profile = High
Encoder control-rate = 1
Encoder EnableTwopassCBR = 0
Opening in BLOCKING MODE 
** Message: 14:31:42.838: <main:4674> iterating capture loop ....
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
^C** Message: 14:36:22.816: <_intr_handler:4262> User Interrupted.. 

Terminating the camera pipeline ...
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
** Message: 14:36:23.368: <main:4684> Capture completed
** Message: 14:36:23.368: <main:4733> Camera application will now exit
```
At least this time, `nvbuf_utils: Could not get EGL display connection` error disappeared. Is there any other possible cause for my issue? Should I reflash my nano device to be sure?

dusty-nv, 2022-07-07:
> There is no display window but there is also no other errors except `[OpenGL] failed to open X11 server connection. [OpenGL] failed to create X11 Window.`

That's okay, that is to be expected when you run it over SSH.

> Could it be something to do the port `1234` in `video-viewer csi://0 rtp://192.168.1.159:1234`. How do I determine the port number in my own system?

Port `1234` is the port on your other system that the RTP stream gets transmitted to.

I don't see why in the second log gstDecoder gets used if the command line is `detectnet csi://0 rtp://192.168.1.159:1234`, normally that would not result in gstDecoder getting used.

closed 2023-04-27 by dusty-nv