dusty-nv/jetson-inference #1438: docker/run.sh fails on xavier nx, manifest unknown

ooijenvangilbert, 2022-06-11:
Hi,

I just got a xavier nx and cloned and build the jetson-inference git as described.
When i try to do the docker/run.sh i get the following output :

ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.7.2
[sudo] password for xavier: 
size of data/networks:  5100826187 bytes
Downloading pytorch-ssd base model...
python/training/det 100%[===================>]  36,23M  10,0MB/s    in 4,2s    
CONTAINER:     dustynv/jetson-inference:r32.7.2
DATA_VOLUME:   --volume /home/xavier/Downloads/jetson-inference/data:/jetson-inference/data --volume /home/xavier/Downloads/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/xavier/Downloads/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/xavier/Downloads/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/xavier/Downloads/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
Unable to find image 'dustynv/jetson-inference:r32.7.2' locally
docker: Error response from daemon: manifest for dustynv/jetson-inference:r32.7.2 not found: manifest unknown: manifest unknown.
See 'docker run --help'.

is my L4T verstion to new? How can i fix this? What am i doing wrong?

Thanks in advance

hey-kong, 2022-06-13:
I have the same problem. I can't get version r32.7.2 from dockerhub, so I temporarily modified `jetson-inference/containers scripts/l4t_version.sh` and fixed L4T_VERSION to r32.7.1.

ooijenvangilbert, 2022-06-13:
Thanks, will have to see when I can test this but will give you feedback once done.

ooijenvangilbert, 2022-06-15:
Hi @dusty-nv 
i got an error saying it could not find the r32.7.0 version. I've changed this to 32.7.1 in the tag.sh file and now it's downloading

dusty-nv, 2022-06-15:
Ahh sorry, thank you, I forgot that R32.7 started with R32.7.1 and not 32.7.0.   Changed this in https://github.com/dusty-nv/jetson-inference/commit/a4f620d4e65f58cc9256e5d1d54855bd5cc14979

closed 2023-04-27 by dusty-nv