dusty-nv/jetson-inference #1439: What is a proper way to increase the accuracy of custom detectnet?

AK51, 2022-06-11:
Dear Dusty,

I have create my custom detectnet, it works ok, thx.  But I want to know the proper way to improve the accuracy.
I use the same 1-minute.mp4 for training and to run the detectnet for checking.

Something I don't understand is that if there is a frame that the object cannot be recognized, I wrote down the time. I then video-capture that object at that time and put in the dataset for re-training. And I did delete the old folder from model folder in order to have a fresh start. (To be sure the engine file is gone.)

But when I run the detectnet again, that object still cannot be recognized. Is there any suggestion to recognize that object at that time?
Note: I did carefully do the camera-capture and there is no wrong labelling and wrong box or empty box. (That is only one label)
Note: The Loss is around 2.6 with 30 epoch
Note: when I use the video-capture, I use merge set option (merge train, val and test)
Thx

AK51, 2022-06-11:
Dear Dusty,

I did it again with more pictures. It is better now as the "exact" time is not exact.
And I am reading
https://github.com/dusty-nv/jetson-inference/issues/1335
If I want to use the previous model, here is the procedure, right?

python3 train_ssd.py --dataset-type=voc --data=data/$NET --model-dir=models/$NET --pretrained_ssd=xxxxx.pth
And delete the engine file in my model folder.

Thx

dusty-nv, 2023-04-27:
@AK51 sorry for the delay - since this post, the `--validate-mean-ap` flag has been added to train_ssd.py which will print out per-class accuracies during the validation phase after each epoch (in addition to being able to run eval_ssd.py).   If one of your classes has lower accuracy, it is recommended to add more training samples for it or re-balance your dataset.

closed 2023-04-27 by dusty-nv