dusty-nv/jetson-inference #1445: Drive AGX Xavier Support

osholsonel20, 2022-06-21:
I understand this doesn't have "Jetpack" on the device, but would there be some ways of modifying the codebase to support running on a drive device? Having trouble integrating the built in driveworks inference solutions so i'm looking into this instead

dusty-nv, 2022-06-21:
Hi @osholsonel20, I haven't developed for the DRIVE platform before, so while I do believe some aspects are similar, I'm not sure of the changes required to support it.

osholsonel20, 2022-06-21:
Sounds good, it has TensorRT 5.1 and Cuda-10.2. Do you know a branch that would suffice this req? I think the only hiccup I've had when running through the install directions was it really wanting me to have jetpack on the system. So if you have any ideas of how to "de-couple" that so it's just utilizing the relevant TensorRT version that'd work too!

closed 2023-04-27 by dusty-nv