dusty-nv/jetson-inference #1448: not mounting cuda in containers when building/running

chris-volley, 2022-06-28:
I have reflashed a couple of times, 

It is not mounting properly the /usr/local/cuda/lib64 into my container. 

My version is 
# R32 (release), REVISION: 7.1, GCID: 29818004, BOARD: t186ref, EABI: aarch64, DATE: Sat Feb 19 17:07:00 UTC 2022

I am trying this command 
docker run -it --rm --runtime nvidia --gpus=all --network host nvcr.io/nvidia/l4t-ml:r32.7.1-py3

If I mount the volume to /usr/local/cuda things work in that container. Any ideas?

dusty-nv, 2022-06-28:
Hi @chris-volley, when you reflashed, did you do it with SDK Manager and did it install the NVIDIA Container Runtime for you, or did you do that manually?

Do you have these CSV files on your system?

``` bash
$ ls -ll /etc/nvidia-container-runtime/host-files-for-container.d/
total 32
-rw-r--r-- 1 root root    26 May 23  2021 cuda.csv
-rw-r--r-- 1 root root  4250 Jul 13  2021 cudnn.csv
-rw-r--r-- 1 root root 12240 Feb  2 16:30 l4t.csv
-rw-r--r-- 1 root root  1590 Jan 14 04:44 tensorrt.csv
-rw-r--r-- 1 root root   325 Aug 11  2020 visionworks.csv
```

dusty-nv, 2022-06-28:
Are you able to run this?

``` bash
$ sudo docker run -it --rm --net=host --runtime nvidia nvcr.io/nvidia/l4t-base:r34.1
# python3 -c 'import tensorrt'
```

I wonder if you had done an `apt-get update && apt-get upgrade` or something after flashing, and if that is messing up your container runtime perhaps?

dusty-nv, 2022-06-28:
Hmm, ok.  Sorry there was a typo in the command above:

``` bash
$ sudo docker run -it --rm --net=host --runtime nvidia nvcr.io/nvidia/l4t-base:r32.7.1
# python3 -c 'import tensorrt'
```

chris-volley, 2022-06-28:
Inside the container all it contains in the /usr/local/cuda/ directory is 
```
ls /usr/local/cuda/lib64
libcudadevrt.a  libcudart_static.a  stubs
```

I assume it's just failing to mount things in the container.

chris-volley, 2022-06-28:
This command allows things to work while running the container.

`sudo docker run -it -v /usr/local/cuda:/usr/local/cuda --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-base:r32.7.1
`
But doesn't solve my problem of making things available when building images.

chris-volley, 2022-06-28:
So I got past this for runtime, 

Not sure why this was the problem, But I have been mounting a separate drive with /home with my stuff in it due to the small amount of space where the os is installed (this persisted after reflashing). Something in there must have been mapped wrong. This time created a new user, and under that user, the nvidia runtime is now loading all the libraries. 

Now I still run into the issue while building that the nvidia runtime is not fully loaded and still has only the same few directories

chris-volley, 2022-06-29:
Thanks, @dusty-nv for your help, I was able to solve this. 

Because of space limitations on the disk, I had created a partition and symlinked /usr/local where cuda is installed to another drive. (because of symlinking I didn't think this would be a problem). However, This was not compatible with the file /etc/nvidia-container-runtime/host-files-for-container.d/cuda.csv. I had to modify the file to the actual location and then it loaded properly into the runtime environment.

andresgalaviz, 2022-07-08:
I had the same issue, it looks like the libraries are located in /usr/local/cuda/lib64/stubs

closed 2023-04-27 by dusty-nv