dusty-nv/jetson-inference #1449: no glDisplay output

zeesee, 2022-06-29:
Good Day!

I'm attempting to command a quick test of detection from a USB camera feed, and the results (live stream overlaid with the bounding boxes of the detected objects) are supposed to be displayed in the OpenGL window; but there's no OpenGL window after running the command.

@dusty-nv Any suggestions or advice would be greatly appreciated. I've spent hours trying to understand and solve this problem, but to no avail.

--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

root@ubuntu:/jetson-inference# detectnet /dev/video0
[gstreamer] initialized gstreamer, version 1.16.2.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: GC21 Video: GC21 Video
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"GC21\ Video:\ GC21\ Video", v4l2.device.bus_info=(string)usb-3610000.xhci-4.3.1, v4l2.device.version=(uint)330305, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 3 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [1] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1 };
[gstreamer] [2] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1 };
[gstreamer] gstCamera -- selected device profile:  codec=mjpeg format=unknown width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! image/jpeg, width=(int)1280, height=(int)720 ! jpegdec ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      mjpeg
  -- width:      1280
  -- height:     720
  -- frameRate:  60.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1920
  -- height:     1080
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]    TensorRT version 8.4.0
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
.
.
.
[TRT]    Tactic: 1000 Time: 0.045964
[TRT]    Tactic: 1002 Time: 0.025911
[TRT]    Tactic: 0 Time: 0.046236
[TRT]    Fastest Tactic: 1002 Time: 0.025911
[TRT]    *************** Autotuning Reformat: Float(135000,1:4,1800,24) -> Float(540000,5625,75,1) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6 -> <out>) (Reformat)
[TRT]    Tactic: 1000 Time: 0.0695723
[TRT]    Tactic: 1002 Time: 0.053888
[TRT]    Tactic: 0 Time: 0.0692523
[TRT]    Fastest Tactic: 1002 Time: 0.053888
[TRT]    *************** Autotuning Reformat: Float(135000,1:4,1800,24) -> Float(540000,1,7200,96) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6 -> <out>) (Reformat)
[TRT]    Tactic: 1000 Time: 0.051075
[TRT]    Tactic: 1002 Time: 0.0451373
[TRT]    Tactic: 0 Time: 0.0506499
[TRT]    Fastest Tactic: 1002 Time: 0.0451373
[TRT]    *************** Autotuning Reformat: Float(135000,1:4,1800,24) -> Half(540000,5625,75,1) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6 -> <out>) (Reformat)
[TRT]    Tactic: 1000 Time: 0.0683392
[TRT]    Tactic: 1002 Time: 0.040141
[TRT]    Tactic: 0 Time: 0.067584
[TRT]    Fastest Tactic: 1002 Time: 0.040141
[TRT]    *************** Autotuning Reformat: Float(135000,1:4,1800,24) -> Half(270000,5625:2,75,1) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6 -> <out>) (Reformat)
[TRT]    Tactic: 1000 Time: 0.0703637
[TRT]    Tactic: 1002 Time: 0.0502263
[TRT]    Tactic: 0 Time: 0.0704299
[TRT]    Fastest Tactic: 1002 Time: 0.0502263
[TRT]    *************** Autotuning Reformat: Float(135000,1:4,1800,24) -> Half(67500,1:8,900,12) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6 -> <out>) (Reformat)
[TRT]    Tactic: 1000 Time: 0.0551253
[TRT]    Tactic: 1002 Time: 0.0264796
[TRT]    Tactic: 0 Time: 0.0532587
[TRT]    Fastest Tactic: 1002 Time: 0.0264796
[TRT]    *************** Autotuning Reformat: Float(135000,1:4,1800,24) -> Half(33750,1:16,450,6) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6 -> <out>) (Reformat)
[TRT]    Tactic: 1000 Time: 0.0541288
[TRT]    Tactic: 1002 Time: 0.0261637
[TRT]    Tactic: 0 Time: 0.054083
[TRT]    Fastest Tactic: 1002 Time: 0.0261637
[TRT]    *************** Autotuning Reformat: Half(540000,5625,75,1) -> Float(540000,5625,75,1) ***************
[TRT]    --------------- Timing Runner: Optimizer Reformat(FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6 -> <out>) (Reformat)
[TRT]    Tactic: 1000 Time: 0.0219073
^Z
[7]+  Stopped                 detectnet /dev/video0

dusty-nv, 2022-06-29:
Hi @zeesee, the first time you run a model, TensorRT takes a few minutes to optimize the model.   So just let this run for a while until it completes.  The next time you start the program it will load in just a few seconds then.

ligaz, 2022-06-30:
You can also generate the engine file before hand using the **trtexec** utility:
```
/usr/src/tensorrt/bin/trtexec --onnx=my_model.onnx --saveEngine=my_model.engine
```

zeesee, 2022-06-30:
understood, thank you! :)

closed 2022-06-30 by zeesee