dusty-nv/jetson-inference #1455: remote host can't get the rtp stream

Im-JimmyHu, 2022-07-13:
1. I convert the usb camera intput on the xvaier to rtsp stream,like this:
```shell
cvlc -vvv v4l2:///dev/video0 --sout '#transcode{vcodec=h264,vb=800,acodec=mp4a}:rtp{sdp=rtsp://192.168.1.82:8554/live.ts}' -I dummy
```
192.168.1.82 is the ip of xvaier.and i use the “rtsp://192.168.1.82:8554/live.ts” as url1

2. then i use the url1 as the input of detectnet ,and push the inference result to rtp steam ,like this
```shell
detectnet rtsp://192.168.1.82:8554/live.ts rtp:192.168.1.16:1234
##192.168.1.16 is the ip of remote host
```
i open the vlc to view "rtp:192.168.1.16:1234" on the remote host.however, the vlc is stuck after geting a frame.so i read the log the tetectnet print.
my error simliar to [this](https://forums.developer.nvidia.com/t/nvbuf-utils-nvreleasefd-failed-exiting/210098),but it not help me.

the complete log of a frame :

> 1 objects detected
> detected obj 0  class #1 (person)  confidence=0.889160
> bounding box 0  (905.625000, 106.523438)  (1893.750000, 1076.835938)  w=988.125000  h=970.312500
> 
> [TRT]    ------------------------------------------------
> [TRT]    Timing Report /data/usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
> [TRT]    ------------------------------------------------
> [TRT]    Pre-Process   CPU   0.04627ms  CUDA   0.12086ms
> [TRT]    Network       CPU  12.59311ms  CUDA  10.94038ms
> [TRT]    Post-Process  CPU   0.01789ms  CUDA   0.00102ms
> [TRT]    Visualize     CPU   0.16637ms  CUDA   1.71008ms
> [TRT]    Total         CPU  12.82365ms  CUDA  12.77235ms
> [TRT]    ------------------------------------------------
> 
> [gstreamer] gstDecoder -- failed to retrieve next image buffer
> detectnet:  failed to capture video frame
> nvbuf_utils: dmabuf_fd 1369 mapped entry NOT found
> nvbuf_utils: NvReleaseFd Failed... Exiting...
> nvbuf_utils: dmabuf_fd 1389 mapped entry NOT found
> nvbuf_utils: NvReleaseFd Failed... Exiting...
> nvbuf_utils: dmabuf_fd 1377 mapped entry NOT found
> nvbuf_utils: NvReleaseFd Failed... Exiting...
> nvbuf_utils: dmabuf_fd 1379 mapped entry NOT found
> nvbuf_utils: NvReleaseFd Failed... Exiting...
> nvbuf_utils: dmabuf_fd 1375 mapped entry NOT found
> nvbuf_utils: NvReleaseFd Failed... Exiting...
> nvbuf_utils: dmabuf_fd 1383 mapped entry NOT found
> nvbuf_utils: NvReleaseFd Failed... Exiting...
> 1 objects detected
> detected obj 0  class #1 (person)  confidence=0.897949
> bounding box 0  (883.125000, 86.748047)  (1895.625000, 1078.945312)  w=1012.500000  h=992.197266
> nvbuf_utils: dmabuf_fd 1385 mapped entry NOT found
> nvbuf_utils: NvReleaseFd Failed... Exiting...
> 
any help would be apprecitaed.

dusty-nv, 2022-07-13:
Can you run `video-viewer rtsp://192.168.1.82:8554/live.ts` and confirm that you can view your RTSP stream with a display attached to your Jetson?

If not, you could try rebuilding jetson-inference with cmake `-DENABLE_NVMM=off` option:

``` bash
cd jetson-inference/build
cmake -DENABLE_NVMM=off ../
make
sudo make install
```

Im-JimmyHu, 2022-07-14:
yes，if i just
```
video-viewer rtsp://192.168.1.82:8554/live.ts
```
although the fps is not good,it's no problem.even i use the  rtsp from remote host(192.168.1.16) as the input:
```
video-viewer rtsp://192.168.1.16:8554/live.ts rtp://192.168.1.16:1234
```
no problem,too.

closed 2023-02-28 by Im-JimmyHu