dusty-nv/jetson-inference #1460: Anybody make it past Posenet on jetson nano 4GB? How about the FPS?

formerlya, 2022-08-02:
I have tried trt_pose on nano,which is 10FPS.
I want to know the FPS about posenet.
Thanks！！

dusty-nv, 2022-08-02:
@formerlya the posenet FPS on Nano should be similar to https://github.com/NVIDIA-AI-IOT/trt_pose#step-3---run-the-example-notebook

formerlya, 2022-08-03:
Thanks for your timely response！
When I runing the model downloader
[image]
Unable to download due to network problem
But in [https://github.com/dusty-nv/jetson-inference/releases](https://github.com/dusty-nv/jetson-inference/releases)
I couldn't find module about pose.
Should I download all of them?

dusty-nv, 2022-08-03:
Hi @formerlya, thanks for pointing these out - I've just uploaded the PoseNet models to that releases page on GitHub, so you should now be able to download them from there.

formerlya, 2022-08-04:
@dusty-nv Thanks!
In my nano, the FPS is 15.
A little faster than [trt_pose](https://github.com/NVIDIA-AI-IOT/trt_pose).
And your answers to other questions have also helped me a lot~

closed 2022-08-04 by formerlya