dusty-nv/jetson-inference #1463: Unable to read MP4 in docker container, what should I do?

wobushihuair, 2022-08-04:
root@wangshuang-HP-Z6-G4-Workstation:/jetson-inference/build/x86_64/bin# video-viewer ../../../test.mp4 
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for ../../../test.mp4
[gstreamer] gstDecoder -- failed to create gstreamer discovery instance:  Couldn't create 'uridecodebin' element
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for file:///jetson-inference/build/x86_64/bin/../../../test.mp4
video-viewer:  failed to create input stream

dusty-nv, 2022-08-04:
It appears that you have built this for x86 - I haven't yet ported gstDecoder to x86, so it will be using gstreamer elements that are only on Jetson and it won't work

wobushihuair, 2022-08-05:
grateful! I am using it on pc for C++ inference of tensorrt, inference of pictures in the container has results but the effect is very poor (using SSD-Mobilenet-v2), and the above error occurs when inferring video.

closed 2023-04-27 by dusty-nv