dusty-nv/jetson-inference #1464: Question about re-training ssd

namaws, 2022-08-05:
Hello, 

I followed the steps to re-train the ssd model with my own dataset and also checked the video tutorial. Does it require to set up the epoch time? And also I wonder why it took me almost three hour to re-train the model with just 200 pictures but on the video it seems pretty fast. 

Thank you

dusty-nv, 2022-08-05:
Hi @namaws, the default number of training epochs with train_ssd.py is 30 if you don't specify it (with the `--epochs` command-line option).  You may need more however if your model isn't accurate enough, and/or add more images to your dataset.

In the youtube video I fast-forwarded btw so the video was not too long.

namaws, 2022-08-05:
Hello @dusty-nv,

Thank you! I'll stop the training and add more images. Is there a mininum number of images you would recommand to have for that?

dusty-nv, 2022-08-05:
IIRC I recommend 100 images per class minimum

namaws, 2022-08-08:
Hello @dusty-nv,

Is it normal to take two hours to just finish epoch 0? (my epoch setup for 5)
thank you

dusty-nv, 2022-08-08:
It shouldn't typically take that long, no.  My guess is that your board is low on memory and is swapping out.  Did you follow these steps?

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

namaws, 2022-08-08:
Hello @dusty-nv,

If I am using jetson nano 2GB developer kit, the swap space can also be 4GB?

dusty-nv, 2022-08-08:
> I've mounted the swap space for 4GB before I trained a few days ago. Do I need to do it everytime before I train?

If you edited `/etc/fstab` like shown in the documentation, then it will be mounted automatically when the system boots

> If I am using jetson nano 2GB developer kit, the swap space can also be 4GB?

You can mount more if you want...I would keep an eye on the memory/swap usage with `sudo tegrastats` to see if you are running low

> Or is it posssible that I didn't really use GPU in jetson nano for training?

It will automatically be used - my guess is that the board is just low memory and swapping out a lot

namaws, 2022-08-08:
Hello @dusty-nv,

Okay thank you so much

mlworkxAI, 2022-09-11:
Hello @dusty-nv 
my code stuck during training not execute at all after few steps
using Nvidia jetson nano 4gb model
1. memory swap already done
2. also desktop gui disabled
but code stil stuck
total images <100
three classes used - grab ,process,place
command used-
 root@mlworkx-desktop:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --dataset-type=voc --data=data/process --model-dir=models/process --batch-size=2 --workers=1 --epochs=1
Thanks in advance

dusty-nv, 2022-09-12:
Hi @namaws, can you try running with `--batch-size=1` instead?   You can also keep an eye on the memory usage with tegrastats.