dusty-nv/jetson-inference #1465: Can't get Jetson imports to run from script file

Unweptpit, 2022-08-08:
I am trying to run a .sh file to launch my python program. Which looks something like this:
```
#!/bin/sh
echo "Starting"
sleep 1
cd /home/velotech/workspace
python3 detect.py
```
And it runs the python program up until it gets to the part where it imports `jetson.inference` and `jetson.utils` when it gets to that point it just closes the terminal. When I run `python3 detect.py` from the terminal manually it works just fine, my program executes normally. It just doesn't work from the .sh file. I don't know what exactly it could be related to, maybe you can help me.

dusty-nv, 2022-08-08:
Hi @Unweptpit, not sure what would be causing this - are you double-clicking this shell script, or running it from terminal?

Is there any output from the program?

What happens if you replace the first line with `#!/usr/bin/env bash` ?

Unweptpit, 2022-08-08:
The situation is as follows, I have a touchscreen attached to a jetson nano. I would like to control everything through the touchscreen (clicking with a mouse). So I've made a .desktop application which runs the .sh script when it is double-clicked. The .sh script then attempts to launch the python program. Which is where the problem occurs. There is no output whatsoever in the terminal, it just launches an empty terminal and closes it immediately. (apart from the first print statement)
I've tried your suggestion, it gave me this error.
[image]

dusty-nv, 2022-08-08:
What if you use `xterm -hold -e "python3 detect.py"` in your script instead?

dusty-nv, 2022-08-09:
Hmm okay - sorry, I don't know what is happening

Can you try making a `.desktop` launcher like shown here:  https://stackoverflow.com/a/23105350

Unweptpit, 2022-08-09:
Thank you for giving me suggestions anyway. I was actually already using a .desktop application to launch the script. I've tried your suggestion and edited the .desktop application to the following.
```
[Desktop Entry]
Name=Velotech
Comment=Run the script in a Terminal window
Exec=bash -c 'cd /home/velotech/workspace && python3 detect.py'
Type=Application
Terminal=true
Icon=utilities-terminal
NoDisplay=true
```
This results in the same behavior as in the beginning, a terminal opens, it prints `this gets printed` and then closes.
Some things I've noticed though, the program runs fine when I use `python3 detect.py` from the terminal myself **but** it also runs fine when I use ./start.sh from the terminal. So the only time it doesn't work is when I double click the .desktop application to launch the script.

Unweptpit, 2022-08-23:
Today I did a complete re-install on my Jetson Nano and did a fresh install of jetson-inference. When I ran the example it threw a warning saying jetson.utils is precated and to use jetson_utils instead. After doing that I was able to run it from a .sh file and by double clicking the .sh file

closed 2022-08-23 by Unweptpit