dusty-nv/jetson-inference #1470: 'jetson.inference.poseNet.ObjectPose.Keypoint' object is not subscriptable

formerlya, 2022-08-17:
When I use the posenet.py.
I want to check if nose/left_eye in pose.Keypoints, and according to its value for further judgment.
①if "nose" in pose.Keypoints:
pose.Keypoints is a list, but it always thought it didn't exist.
#<poseNet.ObjectPose.Keypoint object>
#   -- ID:  0 (nose)
#   -- x:   833.711
#   -- y:   371.706
②if pose.Keypoints[0][0]==nose
TypeError:'jetson.inference.poseNet.ObjectPose.Keypoint' object is not subscriptable.
Tried a lot.

I wonder if you can help me with the list element judgment^_^.
Thank you so much if you can.

formerlya, 2022-08-17:
When I use pose.Keypoints[0].ID, it worked.
But except nose(0) and neck(17), how to judge them if they exist and get their value.
such as left_knee ……

dusty-nv, 2022-08-17:
Hi @formerlya, below you can see a Python example of finding specific keypoints by their name:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/posenet.md#working-with-object-poses

formerlya, 2022-08-18:
Thank you very much for your answer, I am a novice, thank you for your advice!

formerlya, 2022-08-18:
left_wrist_idx = pose.FindKeypoint('left_wrist')

closed 2023-04-27 by dusty-nv