dusty-nv/jetson-inference #1471: Something about videoOutput()

Wadaxivan, 2022-08-17:
I use python to build a program，in this program,I use a CSI camera as a videoSource and “output.mp4” as a videoOutput.But I want to generate several “.mp4” files with running the program once.But it seemed a video can be output only after the program is completed. I want to get the video whenever I want, can it be implemented in the program？(In other words, can I define the end of the video in the program？)

Wadaxivan, 2022-08-17:
Thanks! This may be my problem, I can't find the relevant operation documentation.And I'm sorry to ask when I want to start a new record after I get one video,should I enter a command to start the videoOutput?

Wadaxivan, 2022-08-17:
In addition, is there any official documentation that is more specific about the use of the related functions mentioned above? I may be able to solve some of my low-level problems by myself rather than disturbing you.

dusty-nv, 2022-08-17:
> when I want to start a new record after I get one video,should I enter a command to start the videoOutput?

You should create a new videoOutput object when you want to start writing to a new file, and delete the old one (with `del`)

> is there any official documentation that is more specific about the use of the related functions mentioned above?

Sorry that there isn't great documentation for jetson-utils Python interface - the best thing I would recommend is looking at the jetson-utils .h headers (or the [jetson-utils doxygen](https://github.com/dusty-nv/jetson-inference#api-reference)) and the bindings code under https://github.com/dusty-nv/jetson-utils/blob/master/python/bindings/PyVideo.cpp

Wadaxivan, 2022-08-17:
> You should create a new videoOutput object when you want to start writing to a new file, and delete the old one (with `del`)
> How can I create a new videoOutput?Just use the output.Render(img) can it works?

dusty-nv, 2022-08-17:
You would create a new one like:

```
output = jetson.utils.videoOutput("my_new_video_file.mp4")
```

and then start using that new one instead

Wadaxivan, 2022-08-18:
It works! Thank you!!!

dusty-nv, 2022-08-18:
When you close/delete a videoOutput object, it can take up to a couple seconds to stop gstreamer (but normally shouldn't take that long).   It should be unrelated to the creation of another videoOutput object

Wadaxivan, 2022-08-18:
However, it seems that he did not close successfully, and the generated video cannot be played at this time.

Wadaxivan, 2022-08-18:
In my program, I stop my Output after shot over.
But the pipeline stopped after the new output.
[image]
[image]
[image]
[image]

dusty-nv, 2022-08-18:
Hmm okay...can you try using `del output` instead of `output.Close()`

Also I don't think you should need to re-create the input videoSource, just the videoOutput

Wadaxivan, 2022-08-18:
Thanks! But how can I use the "del output", is this a code?

closed 2023-04-27 by dusty-nv