dusty-nv/jetson-inference #1476: The Imaging Source Camera

Ukhupacha, 2022-08-22:
Hello,

I'm using a specific industrial camera that I connect by USB. Sadly, V4l2 is not working with this camera.
https://github.com/TheImagingSource/tiscamera, this is the repo from the camera.

Could anybody give me some guidance on what to change in order to make v4l2 work with this camera?

I've been digging into the https://github.com/dusty-nv/jetson-utils/blob/master/camera/gstCamera.cpp file in order to make it work but I'm a bit loss.
These are the formats of the camera.

video/x-bayer rggb - 640x480 - 1/1 5/1 15/1 30/1 60/1 120/1 5000000/20833 480/1 601/1 
video/x-bayer rggb - 1440x1080 - 1/1 5/1 15/1 30/1 60/1 120/1 2500000/10593 
video/x-raw BGRx - 640x480 - 1/1 5/1 15/1 30/1 60/1 120/1 5000000/20833 480/1 601/1 
video/x-raw BGRx - 1440x1080 - 1/1 5/1 15/1 30/1 60/1 120/1 2500000/10593 

I would appreciate some advice.

dusty-nv, 2022-08-22:
Hi @Ukhupacha, are you able to view your camera through other V4L2 applications or standalone GStreamer pipelines, or is the issue specific to using programs from jetson-inference/jetson-utils?

```
[gstreamer] gstCamera -- selected device profile: codec=raw format=bayer-rggb width=1440 height=1080
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video1 do-timestamp=true ! video/x-raw, format=(string)rggb, width=(int)1440, height=(int)1080 ! appsink name=mysink
```

This should be a valid pipeline based on your device caps, however it fails to create the pipeline, which makes me think something else is going on in gstreamer or the camera's driver.

Ukhupacha, 2022-08-22:
@dusty-nv,
I'm able to view the camera with the driver they provide that uses V4L2.
[image]

I can also use the gstreamer to view the camera feed with this:
https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/03-live-stream.py

dusty-nv, 2022-08-22:
> gst-launch-1.0 tcambin ! videoconvert ! ximagesink sync=false

That doesn't appear to use the `v4l2src` element, instead it uses their own custom `tcambin` element

You could try modifying jetson-utils to use it too:  https://github.com/dusty-nv/jetson-utils/blob/445c83a2193167d15cfaf6ffb1e620b600be383c/camera/gstCamera.cpp#L166

closed 2023-04-27 by dusty-nv