dusty-nv/jetson-inference #1477: Runtime error (size of tensors) Retraining mobilenet in Colab

obi1brownobi, 2022-08-23:
Good day. I am having an issue when trying to train the ssd-mobilent in colab. I am using the code from dusty-nv/pytorch-ssd. 
When i run this command:
`!python train_ssd.py --dataset-type=voc --data=/content/pytorch-ssd/data/mydata --model-dir=models/myModel --batch-size=32 --num-epochs=30`
I get an error after a few epochs saying:
RuntimeError: Sizes of tensors must match except in dimension 0. Expected size 7 but got size 3 for tensor number 60 in the list.

Its weird because this has worked for me previously and now all of a sudden it won't. Any help would be gladly appreciated.

dusty-nv, 2022-08-23:
Hmm I recall this happening when the number of labels don't match the number of classes in the model

Also does reducing the batch size effect it?

obi1brownobi, 2022-08-23:
Initially i ran the code with a batch size of 4, for 30 epochs. It executed without issues. Now, even if i reduce the batch size to 4, i cannot get past the 5th epoch. The dataset and code are identical to when i first trained successfully.

dusty-nv, 2022-08-23:
It's strange that this runs for 4 epochs and trains over the whole dataset, but then fails during the 5th epoch.  I'm not sure if it's related to the Colab environment or what.  Could you post the error log or exception stack here?

obi1brownobi, 2022-08-24:
I have not altered the code, so I do not see any form of error logs besides for the output displayed in colab, which i have attached here. This time i managed to get the the 9th epoch with the same training command. It's weird that it is failing on a different epoch now...

[training_log.txt](https://github.com/dusty-nv/jetson-inference/files/9413046/training_log.txt)

obi1brownobi, 2022-08-24:
I have included a try/except statement before the quick validation found in the training code. I notice now that I can complete training, although after the first failed quick validation (in whichever epoch this randomly occurs), the remaining epochs fail to validate too. Printing the repeated exception produces the same error described in my first post. I suppose this is not the end of the world, although it would be a nice feature to have at my disposal!

dusty-nv, 2022-08-24:
Thanks for the log - yea, try running train_ssd.py with `--quick-validation` flag to skip that, and it should not encounter the error that triggers the exception

In the meantime I will dig into what is causing the error inside the mAP code (which is a newer addition)

obi1brownobi, 2022-08-24:
Thank you for the help, Sir.

closed 2023-04-27 by dusty-nv