dusty-nv/jetson-inference #1481: Not finding /dev/video0

RoninHunter, 2022-08-24:
My jetson-inference directory does not have a /dev directory even after I plug my camera which is the exact same camera you are using in your walkthrough. I tried to find the /dev/video0 and there is nothing. Was there a step I am missing?

dusty-nv, 2022-08-24:
> My jetson-inference directory does not have a /dev directory

The `/dev` folder isn't under jetson-inference directory, `/dev` is it's own root directory.  Can you try running `ls -ll /dev/video*`

Which camera are you using?   If it's a USB camera, can you run `dmesg` after plugging it in and look for any related errors?

RoninHunter, 2022-08-24:
Logitech c920 is the camera I am using.

When I run both of the suggested commands in the container I get the following.
 
"ls -ll /dev/video*"
ls: cannot access '/dev/video*': No such file or directory

"dmesg or dmesg | more"
dmesg: read kernel buffer failed: Operation not permitted

Outside the container I can see /dev/video0 in the root dir

Also using dmesg -T I can see the usb device getting plugged in without any errors.

dusty-nv, 2022-08-24:
Can you try exiting the container and starting it again after you have plugged in the camera?

If you start the container before plugging in the camera, it won't appear inside the container.

RoninHunter, 2022-08-24:
Ok after restarting the container and plugging in my camera, AFTER I reenter the container, when looking in the /dev directory in root I still see no mounted video0.

RoninHunter, 2022-08-24:
Ok I resolved the issue by having to add the following to the docker run command.

--device=/dev/video0

This allowed to the video0 to show up in the container and now the following command works like a charm. 

video-viewer /dev/video0

Similarly I am still having a problem with the container not seeing my RTSP device. However, I will take a stab at that one before inquire about it in detail. 

Thank you very much though dusty :)

dusty-nv, 2022-08-24:
> Ok I resolved the issue by having to add the following to the docker run command.
> 
> --device=/dev/video0

Oh okay, if you are using the jetson-inference `docker/run.sh` script to start the container, it should already add this for you, but if you are starting the container manually yourself then yes, you would need to add that.  Either way, glad that you got it working!

closed 2023-04-27 by dusty-nv