dusty-nv/jetson-inference #1482: While running "detectnet /dev/video0" video does not display though dectectnet is working

RoninHunter, 2022-08-24:
When running "detectnet /dev/video0" it runs and seems is doing something because the terminal is constantly spitting out [TRT] information but there is not video like in your example. When I run "video-viewer /dev/video0" it sees the camera and opens the video just fine. 

Is there some kind of setting that needs to be toggled to get the video to display when running "detectnet /dev/video0"? What might cause the video to not show while running detect even though it seems to be detecting using the camera?

dusty-nv, 2022-08-24:
Just let detectnet run for awhile (like 5 minutes) - the first time you load a model, TensorRT needs to profile the model and create the engine.  This engine is then cached to disk so the next time you load that model, it will load it just a couple seconds.

RoninHunter, 2022-08-24:
Worked like a charm!!! You rock Dusty!

closed 2023-04-27 by dusty-nv