dusty-nv/jetson-inference #1483: How to process detecnet's ouptut in python

RohanDwivedi, 2022-08-25:
I have managed to retrain ssd mobilenet on custom dataset for xavier nx using this tutorial https://www.forecr.io/blogs/ai-algorithms/how-to-train-ssd-mobilenet-model-for-object-detection-using-pytorch.

The model is working great, now I want to further utilise the detection bounding box coordinates to integrate it with my vision project.  I tried doing as said here https://forums.developer.nvidia.com/t/detectnet-run-external-python-script-if-person-detect/158807, but the changes do not reflect in the code that is running.

I use this command detectnet.py --model=models/dockingStation/ssd-mobilenet.onnx --labels=models/dockingStation/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes /dev/video1 --confidence=0.9 
or 
detectnet --model=models/dockingStation/ssd-mobilenet.onnx --labels=models/dockingStation/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes /dev/video1 --confidence=0.9 

assuming that the first one is python and the other one is c++ and the python changes can be made  to file in examples folder.

dusty-nv, 2022-08-25:
Hi @RohanDwivedi, you need to run a few commands so that the updated detectnet.py gets installed into the correct locations on your system:

```
# after editing jetson-inference/python/examples/detectnet.py
$ cd jetson-inference/build
$ cmake ../
$ make
$ sudo make install
```

Just running the `cmake ../` command will copy ` jetson-inference/python/examples/detectnet.py` to ` jetson-inference/build/aarch64/bin/detectnet.py`.   Running `sudo make install` will install it to `/usr/local/bin`

RohanDwivedi, 2022-08-25:
> Hi @RohanDwivedi, you need to run a few commands so that the updated detectnet.py gets installed into the correct locations on your system:
>
> ```
> # after editing [...]

Hi l, yeah I did exactly this. Just to test I added a print statement and commented out the line that prints performance/timing information and did sudo make install, but it didn't work.

One thing to note, I get an error with sudo not installed so I skipped sudo as the docker is running as root, would that have been the issue!? 

Also, how would I get the coordinates of the detection and change the image being rendered using OpenCV or pillow?

dusty-nv, 2022-08-25:
Oh I see, you are running the Docker container, and editing the file outside of the container.   Is that correct?

What I would do in that case, is make a new folder that contains your edited Python file, and then mount that folder into the container like shown here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-example-python-2.md#setting-up-the-project
```
docker/run.sh --volume ~/my-recognition-python:/my-recognition-python   # mounted inside the container to /my-recognition-python
```

RohanDwivedi, 2022-08-25:
Yeah I am running a docker container! So I use GUI to get to the folder: jetson-inference/python/training/example/  and open the code file in visual studio to edit, save the file and go to my docker terminal to make the file. Is that the wrong way to do it?

I would try mounting the folder and see if that works, alternatively if I run the docker container, cd into the directory, and use nano to open and edit the file, would that work!?

dusty-nv, 2022-08-25:
> Yeah I am running a docker container! So I use GUI to get to the folder: jetson-inference/python/training/example/ and open the code file in visual studio to edit, save the file and go to my docker terminal to make the file. Is that the wrong way to do it?

You are editing the version of the file that exists outside of the container, not inside the container.  This is why you aren't seeing the changes take effect inside the container.  You would essentially need to rebuild the container to get the updated version in there (which is why I recommend just mounting an alternate version)

> I would try mounting the folder and see if that works, alternatively if I run the docker container, cd into the directory, and use nano to open and edit the file, would that work!?

Yes that should work too, just take note of which one you are editing (whether it's the one under `jetson-inference/python/examples`, `jetson-inference/build/aarch64/bin`, or `/usr/local/bin`).  When you exit the container, your changes will be lost however.

RohanDwivedi, 2022-08-25:
> You are editing the version of the file that exists outside of the container, not inside the container.  This is why you aren't seeing the changes take effect inside the container.  You would [...]

Okay, understood. Thanks for the help.

RohanDwivedi, 2022-08-26:
> Oh I see, you are running the Docker container, and editing the file outside of the container.   Is that correct?
>
> What I would do in that case, is make a new folder that contains your edited Python [...]

Hi, so If I copy the ../example/detectNet.py to a new directory, edit the file and mount the directory to the docker. Would I have to run make install  again? Or I can simply cd to the directory and then python3 edited-file.py --

RohanDwivedi, 2022-08-26:
Hi is there a way I could perhaps get ROS working inside this Docker container? I would prefer a method wherein I don't have to lose the data on this container by rebuilding it

closed 2022-08-26 by RohanDwivedi

dusty-nv, 2022-08-26:
> Hi is there a way I could perhaps get ROS working inside this Docker container?

Hi @RohanDwivedi, I have pre-built ROS/ROS2 containers images including PyTorch + jetson-inference up on DockerHub here:

https://hub.docker.com/repository/registry-1.docker.io/dustynv/ros/tags?page=1&ordering=last_updated&name=pytorch

RohanDwivedi, 2022-08-26:
> Hi @RohanDwivedi, I have pre-built ROS/ROS2 containers images including PyTorch + jetson-inference up on DockerHub here: [...]

Oh great. Thanks! I can just copy my .onnx model in the new container and run the model, correct!?

dusty-nv, 2022-08-26:
> Oh great. Thanks! I can just copy my .onnx model in the new container and run the model, correct!?

Yes, or mount it into the container.  It's just like using the jetson-inference container except that it has ROS installed also