dusty-nv/jetson-inference #1484: How to convert ONNX model to tlt

shi093, 2022-08-25:
Hi, Dusty,

I used train_ssd.py to retrain SSD-Mobilenet Model using my own data, and then using onnx_export.py to convert the PyTorch model to ONNX model. The results are really good. However, now I'm trying to use the ONNX model with Deepstream SDK,  and I am struggling to figure out how to convert the ONNX model to tlt model, so that I can use TAO to convert the model into the format that DeepStream can use.

I also tried to convert the ONNX model using Triton server, but failed. Triton can convert ONNX model to TensorRT, but it didn't work for the ONNX model generated from onnx_export.py (it did work on other .onnx model though). So maybe the .onnx model output from onnx_export.py is different from standard onnx model?

I am also wondering why the PyTorch models generated from train_ssd.py is a folder (with .pth) instead of .pt file? 

Thank you so much for your help.

dusty-nv, 2022-08-25:
Hi @shi093, I haven't done this, but I believe that DeepStream can take the ONNX model directly without needing to convert it to a TLT model first.  Here is a thread from another person deploying the SSD-Mobilenet ONNX with DeepStream:   https://forums.developer.nvidia.com/t/deepstream-5-1-pytorch-mobilenet-ssd-v1-retained-onnx-poor-performance/175468

> I am also wondering why the PyTorch models generated from train_ssd.py is a folder (with .pth) instead of .pt file?

I don't think there's really any reason for that - PyTorch doesn't have an official file extension for it's checkpoints, and I've seen `.pth`, `.pt`, `.pth.tar` all used before.  It's just what the code names the checkpoint file as.

shi093, 2022-08-25:
Thank you very much for your reply.
I tried the method of the above link, but didn't work for me. After clone that github repo, and run "make", didn't go through. Maybe that code only works for Jetson platform? 

I think I should have mentioned earlier that I am trying to use the ONNX model trained on Jetson to run on dGPU (V100). Could this be a problem?
Going back to Triton, when I load the ONNX model (output from onnx_export.py) to Triton server, I need to specify a config.pbtxt file, looks like this:

name: "simple-onnx-model"
platform: "onnxruntime_onnx"
max_batch_size: 32
input [
 {
    name: "input_0"
    data_type: TYPE_FP32
    format: FORMAT_NCHW
    dims: [ 3, 224, 224 ]
  }
]
output {
    name: "output1"
    data_type: TYPE_FP32
    dims: [ 1000]
  }

What is the format and dims from the custom ONNX model? as Triton server detected the input dims should be [1, 3, 300, 300] which does not fit NCHW format.

dusty-nv, 2022-08-26:
> I think I should have mentioned earlier that I am trying to use the ONNX model trained on Jetson to run on dGPU (V100). Could this be a problem?

I don't running the ONNX on dGPU is any issue (ONNX models are rather portable), its that I haven't run that model through Triton/DeepStream before so I don't have experience doing that.

> What is the format and dims from the custom ONNX model? as Triton server detected the input dims should be [1, 3, 300, 300] which does not fit NCHW format.

`[1, 3, 300, 300]` is indeed NCHW format, I think you should change `[ 3, 224, 224 ]` to `[1, 3, 300, 300]` because the model was trained with 3x300x300 resolution.

Also there are other things that need changed in this, like the ssd-mobilenet.onnx has multiple output layers.  Sorry that I am of limited assistance when it comes to Triton server.

shi093, 2022-08-26:
Dusty,

Thanks again for your response. I still cannot figure out how to use my customized onnx model in either TAO or Deepstream. So I am now trying to use TAO to retrain a TrafficCamNet model and a DetectNet_v2:resnet34 model using my own data. 

I don't know how you set the parameters in the model training config file, so that after training the mobileNet v2 model using train_ssd.py, the retrained model is so much better than the base model. However, after I retrained the TrafficCamNet model and DetectNet_v2 model, the results from the retrained model got much worse than the original model. There are too many false positives (i.e., detected an object where there is nothing). 

Could you please enlighten me on how you set the training parameters? 

Also, do you have a tlt version of the mobileNet v2 model you used in Hello AI World tutorial (the PyTorch file you asked users to download)? 

Thanks again.

dusty-nv, 2022-08-26:
Hi @shi093, sorry that you are having trouble and that I don't know how to get it converted to TAO.  

I recommend that you post your question to either the Triton Github issues, or the TAO forums:

https://forums.developer.nvidia.com/c/accelerated-computing/intelligent-video-analytics/tao-toolkit/17
https://github.com/triton-inference-server

shi093, 2022-08-29:
Hi, Dusty,

I followed your advice and asked the Triton question on Triton github. After I asked the following question, Triton github host asked me this question about the onnx model: "This looks to be an issue with the model, not serving it in Triton. Does the model work in ONNXRuntime?" So does onnx model generated from "Hello AI World" tutorial work in ONNXRuntime?

dusty-nv, 2022-08-29:
I haven't used ONNXRuntime, but I don't see why it wouldn't work.
It appeared to me the issue was that you were using a config file for an image classifier but this is an object detection model.  It has multiple outputs and requires additional post-processing (like clustering the bounding boxes)

closed 2023-04-27 by dusty-nv