dusty-nv/jetson-inference #1486: Support Jetpack 5.0.2?

kwokkawai, 2022-08-27:
May I know if any docker version supports jetpack 5.0.2?

dusty-nv, 2022-08-28:
Yes there are jetson-inference container for JP 5.0.2 on dockerhub.  You should just be able to do docker/run.sh and it will pull it and run it for you

kwokkawai, 2022-08-28:
thanks, may i know if the docker used in this training have updated too?

https://catalog.ngc.nvidia.com/orgs/nvidia/teams/dli/containers/dli-nano-ai
DLI Getting Started with AI on Jetson Nano | NVIDIA NGC<https://catalog.ngc.nvidia.com/orgs/nvidia/teams/dli/containers/dli-nano-ai>
Course environment for the Deep Learning Institute (DLI) course, "Getting Started with AI on Jetson Nano".
catalog.ngc.nvidia.com

closed 2023-04-27 by dusty-nv