dusty-nv/jetson-inference #1488: Git --recursive problems

Abner-01, 2022-08-31:
When I try to clone recursive the repository or when I use the option `git submodule update --init` I get:

``` remote error: upload-pack: not our ref 8e83c6f0ff5f0c7af5593561d38f07dd6e7bd254
fatal: The remote end hung up unexpectedly
Fetched in submodule path 'python/training/detection', but it did not contain 8e83c6f0ff5f0c7af5593561d38f07dd6e7bd254. Direct fetching of that commit failed.
``
If I continue to build from source, I get the error `-- NumPy not found` . I've alredy installed numpy.

One-star11, 2022-08-31:
I believe the changes made in 12hrs ago is pointing the submodule into a wrong link. I am trying to git in the older version.. hoping it to work..

dusty-nv, 2022-08-31:
Sorry about that @Abner-01 @One-star11 - I'd forgotten to `git push` the `python/training/detection` submodule - my bad.

Please try pulling jetson-inference master again now and it should be working.

closed 2023-04-27 by dusty-nv