dusty-nv/jetson-inference #1489: detectnet works in headless only

RoninHunter, 2022-08-31:
I am able to run detectnet in headless mode and it is working with the usb logitech HD 1080p camera just fine. It detects and prints those detections to the terminal. 

However, when not in headless mode it cannot display the detetions graphically. When I try to do this is starts just fine just like in headless mode until it gets to the following three lines in the terminal output. 

"[OpenGL] glDisplay -- set the window size to 640x480"
"[OpenGL] creating 640x480 texture (GL_RGB8 format, 921600 bytes)"
"Segmentation fault (core dumped)"

Similarly when trying to us video-viewer fails in the same exact spot in the same way, however prior to that it gives the following error.

"[gstreamer] gstDecoder -- failed to retrieve next image buffer"
"video-viewer: failed to capture video frame"

That said, when I run the following gstreamer command it sees the webcam and displays it from within the container smoothly with no problems.

"gst-launch-1.0 v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=30/1' ! autovideosink"

Is there something I need to change within the videoSource or videoOutput to make this work or is there something else I need to do?

As I said, cameras is seen and used in detectnet on headless it just cant open and show the detections output without the above error.

RoninHunter, 2022-08-31:
Interestingly when I go into the python examples I can successfully run gl-display-test.py, however I get the exact same issue described above when running video-viewer.py

dusty-nv, 2022-08-31:
Hi @RoninHunter, are you viewing this from a display physically connected to the Jetson?

The video-viewer/detectnet/ect display won't work over SSH with X11-forwarding because it uses CUDA/OpenGL interoperability.

RoninHunter, 2022-08-31:
Yes, the monitor is connected directly. I can see the camera just fine when using 
"gst-launch-1.0 v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=30/1' ! autovideosink" 

and gl-display-test.py opens a green window and show fast fps at the top

dusty-nv, 2022-08-31:
OK gotcha - which Jetson and version of JetPack-L4T are you using?

RoninHunter, 2022-09-01:
I am using your x86_64 image or your 22.06 image. EVERYTHING is working like a charm accept my ability to display the detection outputs using specifically the following functions

**display.RenderOnce(Image, width, height)**  or  **jetson.utils.glDisplay**

I believe this is where it is breaking.

Opencv will output the video camera and gst-launch-1.0 will output the camera with no problems. 

I guess what would help me the most is pointing me to where in the jetson-inference dir I can see the code of the above two functions are written so that I might be able to get them to work properly.  The python version and the c++ version.

dusty-nv, 2022-09-01:
Oh I see - I don't recall testing the OpenGL display functionality in the x86 container yet (that container is an undocumented "beta" feature still).   I will make a note to test it though, thanks for letting me know!

Here are the source locations btw:

* https://github.com/dusty-nv/jetson-utils/blob/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c/display/glDisplay.cpp#L621

RoninHunter, 2022-09-01:
Does the python examples use the c++ boiler as well or is there a python version of this hiding about somewhere as well?

dusty-nv, 2022-09-01:
The Python bindings just pass it through to the C++ version:

https://github.com/dusty-nv/jetson-utils/blob/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c/python/bindings/PyGL.cpp#L189

RoninHunter, 2022-09-02:
Something is breaking down in the following function within the glTexture.cpp

bool glTexture::init( uint32_t width, uint32_t height, uint32_t format, void* data )

Specifically something in the below is causing a seg fault

`	GL_VERIFY(glTexImage2D(GL_TEXTURE_2D, 0, format, width, height, 0, glTextureLayout(format), glTextureType(format), data));
	
	mID     = id;
	mWidth  = width;
	mHeight = height;
	mFormat = format;
	mSize   = size;

	GL(glBindTexture(GL_TEXTURE_2D, 0));
	GL(glDisable(GL_TEXTURE_2D));`

Is this something that might get looked at in the near future? 
Because if it can get resolved I have a client that will pick up a lot of these Jetson Xaviers.

dusty-nv, 2022-09-02:
> Is this something that might get looked at in the near future?
> Because if it can get resolved I have a client that will pick up a lot of these Jetson Xaviers.

It's on my TODO list to look at in the x86 container, but the x86 version of jetson-inference isn't my priority (the aarch64 version for Jetson is).   This function and the display work fine on Jetson Xavier.  Or is it a requirement that you have x86 support as well?

RoninHunter, 2022-09-02:
"Or is it a requirement that you have x86 support as well?"

**Yes**, we already have a Jetson working like a champ thanks to you. 

We are just trying to set up a mock Jetson environment using x86. 
The x86 environment works fully accept for the above.

dusty-nv, 2022-09-08:
I tried tracking this down, in my case it is crashing in `glTexture::allocInterop()` in the `cudaGraphicsGLRegisterBuffer()` call, and I've been unable to figure out why.   Are you sure it's crashing in `glTexture::init()` in your case?

dusty-nv, 2022-09-08:
@RoninHunter OK - I was able to get this working over here on my x86 system.   These were the changes needed:

* https://github.com/dusty-nv/jetson-inference/commit/7dddf52df57864d00007a8d0ad69ed63fdf2ba35
* https://github.com/dusty-nv/jetson-utils/commit/fad13c48fd864afaa169072d9378b2d0192d5d32

If you clone the jetson-inference `dev` branch (these changes are currently staged in `dev`) and re-build the x86 container, can you let me know if it works?    I tried video-viewer and detectnet on some video files and V4L2 webcam, and it was working fine with the display.

RoninHunter, 2022-09-12:
Hey just saw you replied. I will do just as you say and let you know how it went. THANK YOU

RoninHunter, 2022-09-29:
> @RoninHunter OK - I was able to get this working over here on my x86 system. These were the changes needed:
> 
> * [7dddf52](https://github.com/dusty-nv/jetson-inference/commit/7dddf52df57864d00007a8d0ad69ed63fdf2ba35)
> * [dusty-nv/jetson-utils@fad13c4](https://github.com/dusty-nv/jetson-utils/commit/fad13c48fd864afaa169072d9378b2d0192d5d32)
> 
> If you clone the jetson-inference `dev` branch (these changes are currently staged in `dev`) and re-build the x86 container, can you let me know if it works? I tried video-viewer and detectnet on some video files and V4L2 webcam, and it was working fine with the display.

I went back and did exactly what you have above and it still does not work. Also I should add that the container does not even recognize my camera when doing detectnet or video-viewer unless I install GStreamer using the following command.

`apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio`

Once the above is installed it will run detectnet until it gets to the exact same spot and give the same 

[OpenGL] creating 1280x720 texture (GL_RGB8 format, 2764800 bytes)
Segmentation fault (core dumped)

My run.sh file is as you have it in the updated version and I added the above lines to the glBuffer.cpp in display and rebuilt the container and it still fails for me. 

Could you possibly make a new container with the changes perhaps?

Also, when you do the run.sh file and get into the container do you have to install gstreamer?

dusty-nv, 2022-09-30:
OK, I've updated the `dustynv/jetson-inference:22.06` image on dockerhub with these changes.   I wonder if you weren't building the latest code from the dev branch?

I tested that I'm still able to view my laptop's webcam with the video-viewer and detectnet programs, and they are working fine here.  Also I don't need to install libgstreamer1.0 because those already get installed.

RoninHunter, 2022-09-30:
"I wonder if you weren't building the latest code from the dev branch?" 
Yes I did exactly as you said and pulled that dev branch and used ran the docker/run.sh from it which included the updates you spoke about then built the docker container after adding the few changed you highlighted. However, I will use the new image and see if that resolves the issue. Thank you again Dusty.

probstmartin, 2022-11-18:
> Hi @RoninHunter, are you viewing this from a display physically connected to the Jetson?
> 
> The video-viewer/detectnet/ect display won't work over SSH with X11-forwarding because it uses CUDA/OpenGL interoperability.

Would like to pick up that statement again: there is noway I get video-viewer, etc. run over X11? So either I have to use a direct connection to a screen or VNC?

probstmartin, 2022-11-21:
@dusty-nv Thanks for your quick reply and guidance towards using RTP. I didn't get this to work yet under OS X and VLC but will try again this week. At least I know that I am on the right track now...

dusty-nv, 2022-11-21:
FWIW I've always had better luck using GStreamer to view RTP streams than VLC, and it appears that you can install GStreamer for OS-X.   I've also been working on a WebRTC server implementation for jetson-utils so that you will just be able to view the video stream from your web browser.

RoninHunter, 2022-12-06:
"@RoninHunter OK - I was able to get this working over here on my x86 system. These were the changes needed:

https://github.com/dusty-nv/jetson-inference/commit/7dddf52df57864d00007a8d0ad69ed63fdf2ba35
https://github.com/dusty-nv/jetson-utils/commit/fad13c48fd864afaa169072d9378b2d0192d5d32
If you clone the jetson-inference dev branch (these changes are currently staged in dev) and re-build the x86 container, can you let me know if it works? I tried video-viewer and detectnet on some video files and V4L2 webcam, and it was working fine with the display."

I am so sorry Dusty. I was meaning to reply to this and let you know that following the above instructions did actually yield a working output just as you were getting on your laptop. What I failed to do was rebuild the container as you instructed. 

After doing so the output worked just fine. 

Thank you and sorry for not replying sooner on this topic.

dusty-nv, 2022-12-06:
OK great!, no worries - glad that you got it working

closed 2023-04-27 by dusty-nv