dusty-nv/jetson-inference #1490: Error installing pytorch

changherng, 2022-09-07:
I am using jetson nano jetpack 4.6.2 ubuntu 22.04 with the default python 3.6.9. After running cmake ../ and selecting to install pytorch for python 3.6, I was able to successfully import torch but when I try to import torchvision I get an error.

python3
Python 3.6.9 (default, Jun 29 2022, 11:45:57) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torchvision
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0+78ed10c-py3.6-linux-aarch64.egg/torchvision/__init__.py", line 6, in <module>
    from torchvision import datasets
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0+78ed10c-py3.6-linux-aarch64.egg/torchvision/datasets/__init__.py", line 1, in <module>
    from .lsun import LSUN, LSUNClass
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0+78ed10c-py3.6-linux-aarch64.egg/torchvision/datasets/lsun.py", line 2, in <module>
    from PIL import Image
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/Pillow-9.2.0-py3.6-linux-aarch64.egg/PIL/Image.py", line 52, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 951, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 894, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1157, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1131, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1112, in _legacy_get_spec
  File "<frozen importlib._bootstrap>", line 441, in spec_from_loader
  File "<frozen importlib._bootstrap_external>", line 544, in spec_from_file_location
  File "/usr/local/lib/python3.6/dist-packages/Pillow-9.2.0-py3.6-linux-aarch64.egg/PIL/_deprecate.py", line 1
SyntaxError: future feature annotations is not defined

How do I fix this please?

dusty-nv, 2022-09-07:
Hi @changherng, try running `pip3 install 'pillow<7'`

I can make no guarantees as Ubuntu 22.04 on Nano isn't official JetPack

changherng, 2022-09-07:
Hello @dusty-nv ,
I tried it and it works now, thank you very much!

closed 2022-09-07 by changherng