dusty-nv/jetson-inference #1491: how  can i get cv mat image　from segnet.py?

horo2016, 2022-09-08:
hi，@dusty
I m using segnet.py ， 
`   # render the output image
    output.Render(buffers.output)
    print('********')
    print((buffers.output.channels))
    print((buffers.output.size))
    print(dir(net))
    print('********')
    cv2.imwrite("test.jpg", img_input) #error here  
    # update the title bar
    output.SetStatus("{:s} | Network {:.0f} FPS".format(opt.network, net.GetNetworkFPS()))`
how could i do get cv mat image , i  have known  buffers is a class

dusty-nv, 2022-09-08:
Hi @horo2016, try using the `jetson.utils.cudaToNumpy()` function:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays

horo2016, 2022-09-09:
i have resolved my problem according to your suggestion ,with thanks to your   response in time,it is a bigger help .

```
    array = jetson.utils.cudaToNumpy(buffers.mask)
    print('********')
    print("\ncudaToNumpy() array:")
    print(type(array))
    print(array.dtype)
    print(array.shape)  # numpy dims will be in (height, width, depth) order
    #print(array)
    cv2.imwrite("test.jpg", array)
    print('********')
```

closed 2022-09-09 by horo2016