dusty-nv/jetson-inference #1492: Error when run docker file Jetpack 5.0, TensorRT 8.4

AndreeaVM, 2022-09-08:
Hi!
I'm trying to run a docker image for Jetpack 5.0 with TensorRT 8.4 on the x86 host.  I'm using pytorch:22.07-py3.
I saw the issue https://github.com/dusty-nv/jetson-inference/issues/1124, but I cannot make the downgrade to another TensorRT version because I want to test how the application works and prepare the setup for an Orin Jetson.
So, I also have this problem when I try to build the jetson-inference.
Please, what should I do?

[ 75%] Linking CXX shared library ../../../x86_64/lib/python/3.8/jetson_utils_python.so
[ 75%] Built target video-viewer
[ 75%] Linking CXX executable ../../../x86_64/bin/camera-viewer
[ 75%] Built target jetson-utils-python-38
[ 75%] Built target camera-viewer
/jetson-inference/c/tensorNet.h(617): error: exception specification for virtual function "tensorNet::Profiler::reportLayerTime" is incompatible with that of overridden function "nvinfer1::IProfiler::reportLayerTime"

/jetson-inference/c/tensorNet.h(586): error: exception specification for virtual function "tensorNet::Logger::log" is incompatible with that of overridden function "nvinfer1::ILogger::log"

2 errors detected in the compilation of "/jetson-inference/c/segNet.cu".
CMake Error at jetson-inference_generated_segNet.cu.o.cmake:279 (message):
Error generating file
/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_segNet.cu.o

make[2]: *** [CMakeFiles/jetson-inference.dir/build.make:84: CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_segNet.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/jetson-inference/c/tensorNet.h(617): error: exception specification for virtual function "tensorNet::Profiler::reportLayerTime" is incompatible with that of overridden function "nvinfer1::IProfiler::reportLayerTime"

/jetson-inference/c/tensorNet.h(586): error: exception specification for virtual function "tensorNet::Logger::log" is incompatible with that of overridden function "nvinfer1::ILogger::log"

[ 75%] Building CXX object utils/display/gl-display-test/CMakeFiles/gl-display-test.dir/gl-display-test.cpp.o
In file included from /jetson-inference/build/x86_64/include/jetson-utils/imageFormat.h:240,
from /jetson-inference/build/x86_64/include/jetson-utils/videoOptions.h:26,
from /jetson-inference/build/x86_64/include/jetson-utils/videoOutput.h:27,
from /jetson-inference/build/x86_64/include/jetson-utils/glDisplay.h:26,
from /jetson-inference/utils/display/gl-display-test/gl-display-test.cpp:23:
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl: In function ‘imageFormat imageFormatFromType()’:
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl:235:1: warning: no return statement in function returning non-void [-Wreturn-type]
235 | }
| ^
[ 76%] Linking CXX executable ../../../x86_64/bin/gl-display-test
2 errors detected in the compilation of "/jetson-inference/c/detectNet.cu".
CMake Error at jetson-inference_generated_detectNet.cu.o.cmake:279 (message):
Error generating file
/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_detectNet.cu.o

make[2]: *** [CMakeFiles/jetson-inference.dir/build.make:77: CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o] Error 1
[ 76%] Built target gl-display-test
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl: In function ‘imageFormat imageFormatFromType()’:
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl:235:1: warning: no return statement in function returning non-void [-Wreturn-type]
235 | }
| ^
make[1]: *** [CMakeFiles/Makefile2:436: CMakeFiles/jetson-inference.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
The command '/bin/sh -c mkdir docs && touch docs/CMakeLists.txt && sed -i 's/nvcaffe_parser/nvparsers/g' CMakeLists.txt && mkdir build && cd build && cmake ../ && pwd && ls -al && make -j${NPROC_BUILD} && make install && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 2

Thank you!

dusty-nv, 2022-09-08:
It would seem that this version of TensorRT has some changed functions that I must not have encountered yet.  You can check to see what has changed and make the modifications to tensorNet.h - they should be relatively minor I believe.

AndreeaVM, 2022-09-12:
Hi,

I did all the updates from your last commits and now it's working.
But, while running jetson-inference app, the working station restarts when the engine model is created. Do you have any idea what can be the cause of this sudden restart?

Thank you!

dusty-nv, 2022-09-12:
Hi @AndreeaVM, since you are on x86 I'm not sure what the issue is.  Typically when a machine suddenly turns off or reboots it is a power supply issue.  Or in this case perhaps it could be some kernel fault - is there any dmesg output from before the reboot?   Unfortunately I don't really know as you aren't running on a Jetson.

AndreeaVM, 2022-09-12:
I don't think it is a power supply issue since with docker version for Jetpack 4.6 works fine...
And no help from dmesg :(

Thank you for the help anyway!

dusty-nv, 2022-09-12:
> with docker version for Jetpack 4.6 works fine...

So is this happening on a Jetson?  If so, which Jetson and which power mode are you using?   Try running in a reduced nvpmodel power mode and see if that helps.  If it does, it's a power supply issue.

closed 2023-04-27 by dusty-nv