dusty-nv/jetson-inference #1493: nvbuf_utils: dmabuf_fd 1294 mapped entry NOT found

nawab-sw, 2022-09-09:
Hi all,
I am developing a Pedestrian detection application on Jetson nano(2GB RAM) which is supposed to process video coming from an Hikvision IP camera stream using rtsp.
In my pipeline, I constantly get a the following errors before the program crashes after maximum of 3 minutes:

[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL
nvbuf_utils: dmabuf_fd 1292 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...
nvbuf_utils: dmabuf_fd 1294 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...
[gstreamer] gstDecoder -- pipeline stopped

kindly help me how to solve. i want to continue image processing and detection

dusty-nv, 2022-09-09:
Hi @nawab-sw, I would recommend rebuilding it with the `-DENABLE_NVMM=off` option to cmake:

``` bash
cd jetson-inference/build
cmake -DENABLE_NVMM=off ../
make
sudo make install
```

nawab-sw, 2022-09-10:
> Hi @nawab-sw, I would recommend rebuilding it with the `-DENABLE_NVMM=off` option to cmake:
> 
> ```shell
> cd jetson-inference/build
> cmake -DENABLE_NVMM=off ../
> make
> sudo make install
> ```

Thank you very much . It's working fine now.

closed 2023-04-27 by dusty-nv