dusty-nv/jetson-inference #1494: Training Custom Object Detection Imageset files

changherng, 2022-09-11:
I am currently training a model to detect different bottles, at the moment in my ImageSets folder there are 4 text files 'train.txt', 'test.txt', 'val.txt' and 'trainval.txt'. I noticed when I started training using train_ssd.py the training dataset takes from my 'trainval.txt' and the validation dataset takes from 'test.txt'. May I know if 'train.txt' and 'val.txt' is still needed and how should I allocate my dataset (train 10%, test 10%, val 10% , trainval 70%)?

changherng, 2022-09-11:
Update, I tried training it using 70% on trainval, 10% on the rest and it was successful but after exporting the model to onnx and running it on terminal using detectnet there are a lot of false positives. May I know if I am allocating the ImageSets wrongly?

changherng, 2022-09-11:
[image]
[image]

It was supposed to detect bottles only but I noticed it still detects the objects from "detectnet /dev/video0" where it still detects other objects such as person, vase, clock, tv but now it detects them and changes the list to Soya and Tea only. (Photo attached of it detecting 'tea' is actually detected as 'clock' when I run "detectnet /dev/video0")

I've tried allocating my Imagesets differently and they all had the same result, I also tested them on Jetson nano and Jetson Tx2.  This is my first time encountering this issue, I've successfully done a gesture recognition in the pass and had no problems. I really need help.

changherng, 2022-09-12:
Update: I did everything again and think time instead of collecting images via phone and annotating it with labelimg I just launched the camera-capture /dev/video0, got 250 for train and 40 for test and val. Trained it at 80 epochs (previously 50) and now it works perfectly. I think I did something wrong using the previous method, but I still don't know how to fix it for future use.

dusty-nv, 2022-09-12:
OK, it sounds like you collected a better dataset that was captured with the actual camera you are using, and trained it for more epochs.

Also, note that if you re-train or re-export your model to ONNX, you should delete the .engine file so that TensorRT re-generates it based on the new model.

Parashurambiradar, 2023-04-07:
Hello Dusty-nv,

I'm facing the same issue that chngerng faced in the thread https://github.com/dusty-nv/jetson-inference/issues/1494. can you please let me know how can solve this issue? I trained 700 images and run 500 epochs and still facing the same issue.
environment: Jetson nano and followed your wiki https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-console-2.md#detecting-objects-from-the-command-line 

"Detecting false object with 90% accuracy".

Any help will be greatly appreciated.
Thanks

dusty-nv, 2023-04-27:
@Parashurambiradar sorry for the delay - can you try running train_ssd.py with `--validate-mean-ap` or run eval_ssd.py?   This will print out the per-class accuracies for you, which will let you know if certain object classes need more/less training data.   If this is still an issue, feel free to re-open it.

closed 2023-04-27 by dusty-nv